FPGAwars/apio #26: Implement apio debug install

Obijuan, 2016-03-25:
The "apio debug install" command will download and install the package [tools-usb-ftdi](https://github.com/bqlabs/tools-usb-ftdi)

This package contains the binaries for the utilities listdevs and find_all. The first one is for listing all the USB devices. The second is for showing the boards with FTDI chips

These binaries are invoke using the apio commands:
* apio debug lsusb  ---> listdevs
* apio debug lsftdi --> find_all

Obijuan, 2016-03-25:
The first package for Linux_x86_64: tools-usb-ftdi-linux_x86_64-1.tar.bz2 has been created. It is located here:

https://github.com/bqlabs/tools-usb-ftdi/releases/tag/v0.1

Obijuan, 2016-03-26:
The package for linux 32 bits (i686) is ready:

[tools-usb-ftdi-linux_i686-1.tar.bz2](https://github.com/bqlabs/tools-usb-ftdi/releases/download/v0.1/tools-usb-ftdi-linux_i686-1.tar.bz2)

Obijuan, 2016-03-26:
The package for windows 32 bits is ready:

[tools-usb-ftdi-windows_i386-1.zip](https://github.com/bqlabs/tools-usb-ftdi/releases/download/v0.1/tools-usb-ftdi-windows_i386-1.zip)

Obijuan, 2016-03-27:
The package for ubuntu phone is ready:

[tools-usb-ftdi-armhf-1.tar.bz2](https://github.com/bqlabs/tools-usb-ftdi/releases/download/v0.1/tools-usb-ftdi-armhf-1.tar.bz2)

Obijuan, 2016-03-27:
No me había acordado de que ya había un comando debug.

Podemos sustituirlo por system:

apio system install  --> instala las herramientas para ver el usb y las placas ftdi

Jesus89, 2016-03-28:
Done: 80b32fef06658327b1a5f18e85e49294a71d3292

The commands are:

apio install --system
apio uninstall --system

closed 2016-03-28 by Jesus89