FPGAwars/apio #36: Testing apio install pio-fpga in Mac

Obijuan, 2016-04-19:
Testing the pio install pio-fpga command from a native MAC machine
Follow this installation instructions:

https://github.com/FPGAwars/Platformio-FPGA/wiki/Platformio-FPGA-wiki-home

It would be great to test it both in platformio cli and ide

Jesus89, 2016-04-19:
Done. All OK!

CLI

[image]

IDE

[image]

Simulation

[image]

closed 2016-04-19 by Jesus89

Obijuan, 2016-04-19:
Amazing!!!!!!!!!  :-)