FPGAwars/apio #39: [Feature] Multiboard support in apio

Obijuan, 2016-04-20:
Currently (apio version 0.1.0.2) only supports boards based on lattice ice40hx 1k and tq144  (ex. Icezum Alhambra, icestick).  For supporting other boards, like nandland go, some flags should be passed to apio when synthesizing.

We should think about how to include this multiboard support.  The approach can be similar to apio: use a command for creating a project which includes a project.json file that is read by apio for synthetizing to the target board

Ex:

apio project icezum

It will generate the project.json file in the current directory (aditionally some template files can also be added, like a icezum.pcf constraint file for example)

The algorithm for building could be something like:
$ apio build

1) If SConstruct file exist: Use it for building. Exit
2) Look for the project.json file in the current dir
3) If project.json does not exist 
  2.1 use the flags for the default board (icezum for example)
  2.2 Build it and exit
4) Read the flags in the project.json:  Size (1k, 8k), Pack (tq144, ct256...), family (iCE40), subfamily (HX, LP)
5) Pass this flags to the default SConstruct file (etither by command or by environmental variable)
6) Build

Task to be done:

1) Define the flags needed
2) Create a new SConstruct file that implements this flags
3) Define the project.json file (or board.json or whatever we call it)
4) Implement the command apio project (or whatever name we use for it). It just generates the platform.json file
5) Implement the new apio builld command

I will learn more about the flags needed. Initially I will create a generic makefile and integrate it into apio as an example:  apio examples -f makefile). This is just a legacy example. It will let us compile easily examples that have already been created (like the ones in the verilog tutorials). Then I will create a generic SConstruct file, compatible with the current one, with the flags defined in the begining.  Using apio init this file can be obtained and the user can set manually the building flags. Once it is working, we can integrate it with the new building system

Obijuan, 2016-04-21:
1) Flags needed:

  FPGA_TYPE.  Values: hx, lp
  FPGA_SIZE. Values: 1k, 8k
  FPGA_PACK. Values: swg16tr, cm36, cm49, cm81, cm121, cm225, qn84,  cb81, cb121, cb132, vq100, tq144, ct256

The boards can be determined by these three flags. For example:

icestick / icezum = {1k, hx, tq144}
nandland go board = {1k, hx, vq100}

In addition, each board has it own constraint file (.pcf)

Obijuan, 2016-04-22:
2) The new SConstruct file already implements the flags FPGA_TYPE, FPGA_SIZE and FPGA_PACK

I do not know how to set them when calling scons

The way I am using the go-board is like this:

apio init
edit manually SConstruct for seting the correct values for the go-board (FPGA_PACK='vq100')
apio build

Now the go-board works!

The next step is to read that flags from the project.json file and pass them to scons

@Jesus89 , i think that the definition of the project.json files (or whatever) and the implementation similar to platformio is a task for you

Obijuan, 2016-04-22:
Here I've found instructions to pass the flags to scons:

http://scons.org/doc/1.2.0/HTML/scons-user/x2361.html

Obijuan, 2016-04-22:
Now the folloging flags can be passed to the SConstruct: fpga_type, fpga_size, fpga_pack
For building for the go-board the process is as follows:

$ apio init     --> get the SContruct file
$ scons fpga_pack=vq100

As an intermediate step while implementing the full multiboard support it would be great to pass that flags through apio build as well:

$ apio build fpga_pack=vq100 

What do you think @Jesus89 ?

Jesus89, 2016-04-22:
I think it could be:

$ apio build --pack vq100

But probably in the future, apio.ini file solution will be better. :)

Obijuan, 2016-04-24:
2) The SConstruct file now supports the board command
scons board=icezum

Supported boards:  icezum, icestick, go-board

I've also added it to the build/upload apio commands:

apio build --board go-board
apio upload --board go-board

The default board is the icestick

Obijuan, 2016-04-24:
3)  The apio.ini file will be a json file. Initially it will only contains the board type:

{
  "board" : "icezum"
}

Later we can use the same format than platformio, but for a very early test I think json is ok

Obijuan, 2016-04-24:
Initial multiboard support:

Ex.

$ apio init --board go-board
$ apio build
$ apio upload

It works!  Error checking not yet implemented

I will close this issue when the error checking is implemented

Obijuan, 2016-04-24:
Done!

closed 2016-04-24 by Obijuan