FPGAwars/apio #46: [Feature] Include a short description when listing the examples

Obijuan, 2016-04-21:
The command:

apio examples -l

Generates a list of all the examples available. It would be great if every example has a short description

Ex. Currentle the output is:

$ apio examples -l

 > Makefile
 > icestick
 > leds
 > wire
[..]

It would be great if the output were something like:

$ apio examples -l

 > Makefile    Generic Makefile for building projects (legacy)
 > icestick     Icestick board constraint file
 > leds           Verilog example for Turning all the leds on (for the icestick / icezum boards)
 > wire           Verilog example on how to describe a simple wire

The question is:  Were can be this short string be stored?

Jesus89, 2016-04-24:
Info file in examples.

Jesus89, 2016-04-24:
It would be a good idea to exclude info file in -d and -f commands?

closed 2016-04-24 by Jesus89

Jesus89, 2016-04-24:
Done. Ingore info file in -f, not in -d.

commit 5abd970b (2016-04-24): Merge pull request #68 from Jesus89/develop

Include info in examples -l. Closes #46
commit 75aa38a4 (2016-04-24): Include info in examples -l. Closes #46