FPGAwars/apio #99: Implement the toolchain for the AGC

Obijuan, 2016-08-16:
Create the toolchain for the AGC (Apollo Guide Computer), multiplatform.  Give support in Apio

It is located in [this repo](https://github.com/avtobiff/virtualagc)

The needed tools are:
* yaYUL --> AGC asembler
* yaAGC --> Simulator
* acc-rom --> Generate the rom.list in Verilog (in the [ACC repo](https://github.com/Obijuan/ACC))

zapta, 2024-12-28:
@Obijuan, is this still applicable?

If so, how does this requests translates to apio features?

zapta, 2025-03-25:
@Obijuan, @cavearr, can we close this issue?

cavearr, 2025-03-25:
I prefer @Obijuan close this , i don't know about it.

Obijuan, 2025-03-25:
Please, do not close it. It is a reminder that apio will integrate in the future tools not only for synthesizing hardware but also for assembling code. The idea is using apio for hardware/software co-design . One of the target is the apolo guide computer (AGC)

zapta, 2025-03-25:
@Obijuan, 

1. Where are the assembler/linker tools of the AGC?

2. How the content of rom.list will loaded into the AGC core (when running on an actual FPGA board)? Will it be baked into the Verilog model or downloaded separately?

3. Did the 'a' in 'apio' come from 'Apollo'?

zapta, 2026-01-19:
This proposed AGC specific toolchain would better be implemented as an independent tool that uses apio for the Veilog -> bitstreams -> upload steps. 

Apio can now easily be reused by higher level tools using its self contains file bundles.

Closing this one.

closed 2026-01-19 by zapta