FPGAwars/apio #114: [Apio 0.1.6.4] The comand "apio examples -l" gives different outputs in windows than in Linux

Obijuan, 2016-09-30:
I am testing Apio 0.1.6.4

When this command is issued:

```
$ apio examples -l
```
what is shown in windows 7 is:

[image]

Instead of the names, you see the full path

But in Linux, what is shown are the names of the examples (not the paths):

[image]

I am not sure if this is a bug in apio for windows. I think that the output should be similar in both systems

Jesus89, 2016-09-30:
Execute `apio install -l`.

You have different versions of _examples_ package.

`apio install examples` will do the job.

Also, I will check this for windows. It should appear _icezum/leds_ instead of the full path.

Obijuan, 2016-10-12:
I can confirm that this bug is fixed. I've tested it on Windows 7 virtual machine. Everything was ok

closed 2016-10-12 by Obijuan