FPGAwars/apio #127: [GNU/Linux] `apio drivers --enable` fails if `sudo` is not installed.

1138-4EB, 2017-03-29:
For example, on Ubuntu without sudo:

```
# apt-get update && apt-get install -y python python-pip \
 && pip install -U pip apio \
 && apio drivers --enable \
 && apio install --all
```

throws the following error:

```
Configure FTDI drivers for FPGA
Traceback (most recent call last):
  File "./root/.local/bin/apio", line 11, in <module>
    sys.exit(cli())
  File "/root/.local/lib/python2.7/site-packages/click/core.py", line 722, in __call__
    return self.main(*args, **kwargs)
  File "/root/.local/lib/python2.7/site-packages/click/core.py", line 697, in main
    rv = self.invoke(ctx)
  File "/root/.local/lib/python2.7/site-packages/click/core.py", line 1066, in invoke
    return _process_result(sub_ctx.command.invoke(sub_ctx))
  File "/root/.local/lib/python2.7/site-packages/click/core.py", line 895, in invoke
    return ctx.invoke(self.callback, **ctx.params)
  File "/root/.local/lib/python2.7/site-packages/click/core.py", line 535, in invoke
    return callback(*args, **kwargs)
  File "/root/.local/lib/python2.7/site-packages/click/decorators.py", line 17, in new_func
    return f(get_current_context(), *args, **kwargs)
  File "/root/.local/lib/python2.7/site-packages/apio/commands/drivers.py", line 24, in cli
    exit_code = Drivers().enable()
  File "/root/.local/lib/python2.7/site-packages/apio/managers/drivers.py", line 52, in enable
    return self._enable_linux()
  File "/root/.local/lib/python2.7/site-packages/apio/managers/drivers.py", line 78, in _enable_linux
    self.rules_local_path, self.rules_system_path])
  File "/usr/lib/python2.7/subprocess.py", line 523, in call
    return Popen(*popenargs, **kwargs).wait()
  File "/usr/lib/python2.7/subprocess.py", line 711, in __init__
    errread, errwrite)
  File "/usr/lib/python2.7/subprocess.py", line 1343, in _execute_child
    raise child_exception
OSError: [Errno 2] No such file or directory
```
I think that the problem might be at `managers/drivers.py`:

```
def _enable_linux(self):
click.secho('Configure FTDI drivers for FPGA')
if not isfile(self.rules_system_path):
subprocess.call(['sudo', 'cp',
self.rules_local_path, self.rules_system_path])
subprocess.call(['sudo', 'service', 'udev', 'restart'])
```

If `apio drivers --enable` is run as root, `sudo` is not required. Furthermore, shall it not be available, trying to use it throws an error. I'd expect apio to check whether `sudo` is required, prior to using it.

---

On top of that, if Ubuntu is a docker container, restarting udev won't work. However, I have not looked in depth into this. So, I don't know which the best approach would be: whether not to run that command and show a message instead, or replace `drivers --enable` as a whole and show a longer message. You can have a look at this question: http://stackoverflow.com/questions/36880565/why-dont-my-udev-rules-work-inside-of-a-running-docker-container

Jesus89, 2017-03-30:
The error says that directory '/etc/udev/rules.d/' does not exist. In order to show a better message I can catch this exception.

About the sudo, I think this solution can be used to cover non-sudo systems: 'cmd 2>/dev/null || sudo cmd'.

I don't think it is a good idea to configure the drivers in a docker container. May be the configuration must be done in the main system to enable FTDI comms. If you succeed with this, let me know.

1138-4EB, 2017-03-30:
> The error says that directory '/etc/udev/rules.d/' does not exist. In order to show a better message I can catch this exception.

I would say that the error is not triggered in lines 1-3, but exactly in the fourth:

```
subprocess.call(['sudo', 'cp',self.rules_local_path, self.rules_system_path])
```
Indeed, ensuring that the directoy actually exists, makes no difference. You can check it yourself.

> About the sudo, I think this solution can be used to cover non-sudo systems: 'cmd 2>/dev/null || sudo cmd'.

Won't `apio drivers --enable 2>/dev/null || sudo apio drivers --enable` execute the exact same code? I.e., `subprocess.call(['sudo', 'cp', self.rules_local_path, self.rules_system_path])`

> I don't think it is a good idea to configure the drivers in a docker container. May be the configuration must be done in the main system to enable FTDI comms. If you succeed with this, let me know.

An interesting feature of docker images is that a command (or sequence of commands) can be executed as soon as a container is started. Therefore, the content in `/etc/udev/rules.d` can be copied from the container to the host and execute then `service udev restart`.

Furthermore, I'd say that `service udev restart` is biased to Ubuntu systems. E.g., in latest Arch/Fedora releases `systemctl restart systemd-udevd.service` would be the preferred command: http://man7.org/linux/man-pages/man8/systemd-udevd.service.8.html Then, allowing to avoid it's execution (or automatically disabling it if a certain environment variable is set), would help to move that step in order to be executed later.

Jesus89, 2017-03-31:
... Please, develop a solution that works in your system and then do a contribution.

closed 2017-03-31 by Jesus89

1138-4EB, 2017-04-01:
> ... Please, develop a solution that works in your system and then do a contribution.

Any of the proposed solutions work in my system. Furthermore, I'm contributing this issue and the thread in the forum.