FPGAwars/apio #128: Package tools in $PATH

aw, 2017-04-24:
Hello,

I've installed `apio` and noticed the tools from packages are not accessible.

Example:

```
$ apio install icestorm
...
$ iceprog
-sh: iceprog: not found
```

After some searching, I did find the tools are located in

    ~/.apio/packages/toolchain-icestorm/bin/

..although I think it would be better if they were automatically added to the user `$PATH`, or installed in `/usr/local/bin/`.

Same problem for `lsusb`, `lsftdi`, etc.

Jesus89, 2017-04-24:
Hi!

All apio packages are installed in the isolated environment "~/.apio" to avoid conflicts with other packages in your system and to provide a package structure compatible with Windows and Mac.

If you want to use directly in your Linux system you can add it manually into the path by adding these lines in your `~/.bashrc` file:

```
export PATH=~/.apio/packages/toolchain-icestorm/bin:$PATH
export PATH=~/.apio/packages/toolchain-iverilog/bin:$PATH
export PATH=~/.apio/packages/tools-system/bin:$PATH
```

Then run `source .bashrc` to apply it. It will be automatically applied to new terminals.

Other solution is to download directly the packages and uncompress them in `/usr/local/`, `~/.local`, etc. These are the package's repositories:

* https://github.com/FPGAwars/toolchain-icestorm/releases
* https://github.com/FPGAwars/toolchain-iverilog/releases
* https://github.com/FPGAwars/tools-system/releases

Regards.

aw, 2017-04-24:
Hello @Jesus89, I didn't realize I can download the packages individually that way. That's perfect.

Thank you!

closed 2017-04-24 by aw