FPGAwars/apio #131: Add support to the Mystorm Blackice board

Obijuan, 2017-08-29:
Main issue for giving support to the mystorm blackice open fpga board. This issue will be break down into smaller issues. This issue will be closed when the initial support is working ok

Roadmap
--------

* Add the blakice in the list of board, so that the build can be done with apio build command (and the .ini apio file)
* Create a python tool for downloading into the board (similar to cat > /dev/ttyUSB). This tool should be multiplatform (Windows, mac, linux). black-iceprog? (a name is needed)
* Include support for auto-detecting the blackice board serial port. I think this is the most difficult part
* Include the blackiceprog toolchain as part of apio (install, uninstall, etc...)
* Create a new scons file that calls blackiceprog instead of iceprog when the blackice board is used
* Implement the apio upload command for the blackice
* Include some examples: blackice/leds, etc...

jpenalbae, 2017-11-06:
After PR #135 the only missing things are examples which I will probably do during next weekend. And maybe creating a package for the the `black-iceprog.py` script and upload it to pypi to be able to set it as dependency.

tinyfpga, 2017-11-06:
@jpenalbae, @Obijuan : I added a basic template example in my pull request for apio-examples: https://github.com/FPGAwars/apio-examples/pull/9

Obijuan, 2017-11-06:
Thanks a lot! This is great!  @Jesus89 and I will test it before closing the issue

Jesus89, 2017-12-13:
Closed by: https://github.com/FPGAwars/apio/pull/139

closed 2017-12-13 by Jesus89