FPGAwars/apio #136: Migration to python 3

Obijuan, 2017-11-06:
Currently apio is based on Python 2.7.  This is because apio depends on [Scons](http://scons.org/), an open source and multiplatform software construction build python. And Scons depends on Python 2.7. It seems that they are migrating it to python 3, but the progress is very slow

There are two options: wait for the migration of scons to python3 or change the building system. It is an open question

What do you think? Comments are welcome :-)

Obijuan, 2017-11-22:
It seems that scons 3.0.1 is now migrated to python 3! :-)

http://scons.org/scons-301-is-available.html

Obijuan, 2017-12-01:
I've tested the new scons 3.0.1 with python3 and it works great!!!!

Here you are the instructions for testing it:

* Create a virualenv with python3
```
virtualenv -p python3 scons-test
```
* Activate the environment:
```
cd scons-test/
source bin/activate
```

* check the python version installed:
```
$ python --version
Python 3.5.2
```

* Install scons
```
pip install scons
```

* check the scons version. It should be 3.0.1 or greather:

```
$ scons --version
SCons by Steven Knight et al.:
	script: v3.0.1.74b2c53bc42290e911b334a6b44f187da698a668, 2017/11/14 13:16:53, by bdbaddog on hpmicrodog
	engine: v3.0.1.74b2c53bc42290e911b334a6b44f187da698a668, 2017/11/14 13:16:53, by bdbaddog on hpmicrodog
	engine path: ['/home/obijuan/develop/virtualenv/scons-test/lib/python3.5/site-packages/scons-3.0.1/SCons']
Copyright (c) 2001 - 2017 The SCons Foundation
```

* Initial test: use whatever verilog proyect you have. Copy the proyect into a folder inside the virtual env. For synthesizing in a icestick or Icezum Alhambra boards use the following command:

```
scons fpga_size=1k fpga_type=hx fpga_pack=tq144
```

* Install apio in the virtualenv:

```
pip install apio
```

* Install the old scons:

```
apio install scons
```

* Download the taz scons 3.0.1 package from [here](http://prdownloads.sourceforge.net/scons/scons-3.0.1.tar.gz) 

* Uncompress the package

* Remove all the files under the $HOME/.apio/packages/tool-scons

* Copy all the files form the scons-3.0.1 under the $HOME/.apio/packages/tool-scons folder

Now you have apio with the new scons 3.0.1

* You can test apio build, apio time, apio sim, etc... with different proyects

Conclusion:  Scons 3.0.1 can be included in apio. We no longer need python 2.7 :-)

This issue will be still open until this integration work is finished

Jesus89, 2017-12-13:
Closed by: https://github.com/FPGAwars/apio/pull/139

closed 2017-12-13 by Jesus89