FPGAwars/apio #143: ImportError: No module named SCons.Script

adumont, 2018-02-04:
Today I'm getting this error when running apio: "ImportError: No module named SCons.Script"

It's been working fine for days, but maybe some apt upgrade/remove... might have messed things up. I haven't been able to find the root cause, not a good fix either. 

I have been able to work around the issue by exporting:

SCONS_LIB_DIR=/usr/local/lib/python2.7/dist-packages/scons-3.0.1

After that, apio works again.

Any idea?

The error: (same happens when running apio clean, apio verify,...)

adumont@denia:~/Robotica/Alex/FPGA/Verilog/Automaton$ apio clean
SCons import failed. Unable to find engine files in:
/usr/local/bin/../engine
/usr/local/bin/scons-local-3.0.1
/usr/local/bin/scons-local
/usr/local/lib/python2.7/dist-packages/lib/scons-3.0.1
/usr/local/lib/scons-3.0.1
/usr/lib/scons-3.0.1
/usr/local/lib/scons-3.0.1
/usr/local/lib/python2.7/dist-packages/lib/python2.7/site-packages/scons-3.0.1
/usr/local/lib/python2.7/site-packages/scons-3.0.1
/usr/lib/python2.7/site-packages/scons-3.0.1
/usr/local/lib/python2.7/site-packages/scons-3.0.1
/usr/lib/scons-3.0.1
/usr/local/lib/python2.7/dist-packages/lib/scons
/usr/local/lib/scons
/usr/lib/scons
/usr/local/lib/scons
/usr/local/lib/python2.7/dist-packages/lib/python2.7/site-packages/scons
/usr/local/lib/python2.7/site-packages/scons
/usr/lib/python2.7/site-packages/scons
/usr/local/lib/python2.7/site-packages/scons
/usr/lib/scons
Traceback (most recent call last):
File "/usr/local/bin/scons", line 192, in <module>
import SCons.Script
ImportError: No module named SCons.Script
============================================= [ ERROR ] Took 0.22 seconds =============================================

If that helps:

adumont@denia:~$ pip install -U apio
Requirement already up-to-date: apio in /usr/local/lib/python2.7/dist-packages
Requirement already up-to-date: colorama in /usr/local/lib/python2.7/dist-packages (from apio)
Requirement already up-to-date: click<7,>=5 in /usr/local/lib/python2.7/dist-packages (from apio)
Requirement already up-to-date: scons==3.0.1 in /usr/local/lib/python2.7/dist-packages (from apio)
Requirement already up-to-date: requests<3,>=2.4.0 in /usr/local/lib/python2.7/dist-packages (from apio)
Requirement already up-to-date: semantic-version>=2.5.0 in /usr/local/lib/python2.7/dist-packages (from apio)
Requirement already up-to-date: pyjwt<2,>=1.5.3 in /usr/local/lib/python2.7/dist-packages (from apio)
Requirement already up-to-date: pyserial<4,>=3 in /usr/local/lib/python2.7/dist-packages (from apio)
Requirement already up-to-date: certifi>=2017.4.17 in /usr/local/lib/python2.7/dist-packages (from requests<3,>=2.4.0->apio)
Requirement already up-to-date: chardet<3.1.0,>=3.0.2 in /usr/local/lib/python2.7/dist-packages (from requests<3,>=2.4.0->apio)
Requirement already up-to-date: idna<2.7,>=2.5 in /usr/local/lib/python2.7/dist-packages (from requests<3,>=2.4.0->apio)
Requirement already up-to-date: urllib3<1.23,>=1.21.1 in /usr/local/lib/python2.7/dist-packages (from requests<3,>=2.4.0->apio)

adumont@denia:~$ pip -V
pip 9.0.1 from /usr/local/lib/python2.7/dist-packages (python 2.7)

Jesus89, 2018-02-17:
Hi!

This issue may be related to: https://github.com/FPGAwars/apio/issues/144.

Please, try to install scons manually `pip install -U scons`. If you have problems update the installing tools:

```
pip install -U setuptools
pip install -U wheel
```

adumont, 2018-02-17:
Hi Jesus, thanks for the reply.

I've tried that:

$ pip install -U scons
Requirement already up-to-date: scons in /usr/local/lib/python2.7/dist-packages

$ pip freeze | grep scons
scons==3.0.1

BTW I'm using : 
$ python -V
Python 2.7.12

I have also installed setuptools and wheel, but same error still happens.

I'm on Apio 0.3.0.

adumont, 2018-02-18:
Tried what @Obijuan says in #146 with no luck. I also uninstalled and reinstalled scons after, but same error still happens.

jjchico, 2018-02-19:
Hi!
Same problem here using Ubuntu 16.04. All packages installed through "pip3".
```
$ cat /etc/issue
Ubuntu 16.04.3 LTS \n \l
$ python --version
Python 2.7.12
$ python3 --version
Python 3.5.2
$ apio --version
apio, version 0.3.0
```
```
$ apio verify
SCons import failed. Unable to find engine files in:
/home/jjchico/.local/bin/../engine
/home/jjchico/.local/bin/scons-local-3.0.1
/home/jjchico/.local/bin/scons-local
/home/jjchico/.local/lib/scons-3.0.1
/usr/lib/scons-3.0.1
/usr/local/lib/scons-3.0.1
/home/jjchico/.local/lib/python2.7/site-packages/scons-3.0.1
/usr/lib/python2.7/site-packages/scons-3.0.1
/usr/local/lib/python2.7/site-packages/scons-3.0.1
/usr/lib/scons-3.0.1
/home/jjchico/.local/lib/scons
/usr/lib/scons
/usr/local/lib/scons
/home/jjchico/.local/lib/python2.7/site-packages/scons
/usr/lib/python2.7/site-packages/scons
/usr/local/lib/python2.7/site-packages/scons
/usr/lib/scons
Traceback (most recent call last):
File "/home/jjchico/.local/bin/scons", line 192, in <module>
import SCons.Script
ImportError: No module named SCons.Script
========================= [ ERROR ] Took 0.25 seconds =========================
```
It looks like the scon executable is only looking for python 2.7 libraries. Note that "python" defaults to 2.7 in Ubuntu 16.04.

### Workarounds

* Define SCONS_LIB_DIR:
```
export SCONS_LIB_DIR=~/.local/lib/python3.5/site-packages/scons-3.0.1
```

* Edit scons executable to run under python3:
```
$ cat ~/.local/bin/scons | grep '^#!'
#! /usr/bin/env python <<< REPLACE WITH 'python3'
```

Jesus89, 2018-03-03:
Hi. The issue is that the `scons` script it's being executed by default with `python 2.7` although it has been installed with `python 3.5`. Son in [this line](https://github.com/SCons/scons/blob/master/src/script/scons.py#L160) it is detecting the version of `2.7` instead of the good one in this case `3.5`.

If you execute the scons script with the correct python version, the issue is solved:

```

Jesus89, 2018-03-03:
Fixed here: 61f5260968152d727245b6711a97626c09b789c3

Could you try to update apio to test the development version?

```
pip install -U git+git://github.com/fpgawars/apio.git
```

Thanks!

jjchico, 2018-03-03:
Yes! This update solves the problem... so far.
Thanks!

closed 2018-03-03 by Jesus89

StefanMavrodiev, 2018-03-21:
Hello, I'm having issue with scons. Maybe it's related to this issue or maybe not.

Anyway,
I'm using python2.7 and ubuntu 16.04

The error is:
```sh
 Executing: scons -Q build fpga_size=1k fpga_type=hx fpga_pack=tq144 -f /usr/local/lib/python2.7/dist-packages/apio-0.3.1-py2.7.egg/apio/resources/SConstruct
 /usr/bin/python: can't open file '/usr/local/lib/python2.7/dist-packages/apio-0.3.1-py2.7.egg/apio/scons': [Errno 2] No such file or directo
```

In util.py:
```python
scons_command = [
    sys.executable,
    join(dirname(APIO_MAIN_FILE), 'scons')
]
```

Clearly the path is somehow wrong.
Why don't just call scons?
```python
scons_command = [
    'scons'
]
```

beyond-wyc, 2020-01-07:
chmod +755 -R /usr/lib/scons-2.4.1/