FPGAwars/apio #144: error: option --single-version-externally-managed not recognized

Obijuan, 2018-02-15:
I am testing apio in a new win10 machine

I've installed python3  (3.6.4, the current latest stable version)

When installing apio with "pip install apio" this is what happens:

[image]

I've copy&pasted the last part:

```
 Running setup.py install for scons ... error

 Complete output from command c:\users\obijuan\appdata\local\programs\python\python36-32\python.exe -u -c "import setuptools, tokenize;__file__='C:\\Users\\obijuan\\AppData\\Local\\Temp\\pip-build-umawyo6t\\scons\\setup.py';f=getattr(tokenize, 'open', open)(__file__);code=f.read().replace('\r\n', '\n');f.close();exec(compile(code, __file__, 'exec'))" install --record C:\Users\obijuan\AppData\Local\Temp\pip-dafibkpg-record\install-record.txt --single-version-externally-managed --compile:
    usage: -c [global_opts] cmd1 [cmd1_opts] [cmd2 [cmd2_opts] ...]
       or: -c --help [cmd1 cmd2 ...]
       or: -c --help-commands
       or: -c cmd --help

    error: option --single-version-externally-managed not recognized
```

And this is the red message:

```
Command "c:\users\obijuan\appdata\local\programs\python\python36-32\python.exe -u -c "import setuptools, tokenize;__file__='C:\\Users\\obijuan\\AppData\\Local\\Temp\\pip-build-umawyo6t\\scons\\setup.py';f=getattr(tokenize, 'open', open)(__file__);code=f.read().replace('\r\n', '\n');f.close();exec(compile(code, __file__, 'exec'))" install --record C:\Users\obijuan\AppData\Local\Temp\pip-dafibkpg-record\install-record.txt --single-version-externally-managed --compile" failed with error code 1 in C:\Users\obijuan\AppData\Local\Temp\pip-build-umawyo6t\scons\
```

Obijuan, 2018-02-15:
There is a problem in the scons packages. The same error is gotten simply trying to install scons...

```
C:\Users\obijuan>pip install scons
Collecting scons
  Using cached scons-3.0.1.tar.gz
Installing collected packages: scons
  Running setup.py install for scons ... error
    Complete output from command c:\users\obijuan\appdata\local\programs\python\python36-32\python.exe -u -c "import setuptools, tokenize;__file__='C:\\Users\\obijuan\\AppData\\Local\\Temp\\pip-build-p_gstqkv\\scons\\setup.py';f=getattr(tokenize, 'open', open)(__file__);code=f.read().replace('\r\n', '\n');f.close();exec(compile(code, __file__, 'exec'))" install --record C:\Users\obijuan\AppData\Local\Temp\pip-d8iraecx-record\install-record.txt --single-version-externally-managed --compile:
    usage: -c [global_opts] cmd1 [cmd1_opts] [cmd2 [cmd2_opts] ...]
       or: -c --help [cmd1 cmd2 ...]
       or: -c --help-commands
       or: -c cmd --help

    error: option --single-version-externally-managed not recognized

    ----------------------------------------
```

I will try to get more info on this

Obijuan, 2018-02-15:
OK. It solves the problem:

```
pip install -U setuptools
pip install -U wheel
```

It is necessary to have the latest setup and wheels tools

closed 2018-02-15 by Obijuan

luxiaolei, 2019-05-12:
Not worked for me..

Braiunito, 2019-06-09:
> Not worked for me...

If not working for you it's possibly what you don't have Microsoft C++ 14.0, It's common that this module will be necessary for any other python modules, for more information, read the follow post: https://www.scivision.dev/python-windows-visual-c-14-required/