FPGAwars/apio #146: [Bug] Installing scons via 'pip install scons' doesn't work

Obijuan, 2018-02-18:
This is not an Apio bug but is reported here because is affecting the apio installation

This issue is comments in the scons user list:  [installing scons via pip install scons doesn't work](http://scons-users.scons.narkive.com/vNSIhTqN/installing-scons-via-pip-install-scons-doesn-t-work)

When trying to install apio with "sudo pip install apio" there is an strange message gotten:  "error: option --single-version-externally-managed not recognized"

The problem is due to the scons package. It can be reproduced just by installling scons itself with:

```
 sudo pip install scons
```

In windows it seems to be fixed just by updating the setuptools and wheel

```
pip install -U setuptools
pip install -U wheel
```

(but more tests should be done)

The solution in linux is to install scons in a virtual environment. Then, if you install apio (inside or outside the virtualenv) it will work

* Install virtualenv

```
sudo apt install virtualenv python3-virtualenv
```

* Create a virtualenv with python3

```
virtualenv -p python3 test
```

* Enter the virutalenv and activate it:

```bash
cd test
source bin/activate
```

* Install scons

```
pip install scons
```

Now you can install apio inside the virtualenv, or outside

Jesus89, 2018-03-03:
I have tested this steps. It works inside the virtualenv but then it is not fixed outside.

The bug is related to the detection of the `libs` in the [main script](https://github.com/SCons/scons/blob/master/src/script/scons.py). This `libs` are added to `sys.path` in order to `import SCons.Script`.

Since the virtualenv has only one version of Python, scons can detect and load the correct paths. Otherwise, if you have different versions of Python installed in your system, it can not detect the correct version for Python 3.5.

A solution is to setup the `SCONS_LIB_DIR` (which it is added to the `libs`) with the environment path of the *current apio version*. This is also good because apio will execute always *the correct scons*, in case there are multiple scons installed in the system.

Jesus89, 2018-03-03:
Since this issue is [duplicated](https://github.com/FPGAwars/apio/issues/143) I can close it and continue there :smile:

closed 2018-03-03 by Jesus89

Obijuan, 2018-03-11:
@Jesus89 found this alternative solution:

https://github.com/platformio/platformio-core/issues/279

It seems that installing scons with the following command does the trick

```
sudo pip install --egg scons
```

physicst, 2019-05-02:
Traceback (most recent call last):
  File "/home/atif/rat/rat-pac-jsns2/bin/scons", line 63, in <module>
    import SCons.Script
  File "/home/atif/rat/rat-pac-jsns2/python/SCons/__init__.py", line 43, in <module>
    import SCons.compat
  File "/home/atif/rat/rat-pac-jsns2/python/SCons/compat/__init__.py", line 178, in <module>
    import shlex
  File "/home/atif/python/python/Python-2.7.6/Lib/shlex.py", line 12, in <module>
    from collections import deque
  File "/home/atif/python/python/Python-2.7.6/Lib/collections.py", line 8, in <module>
    from _collections import deque, defaultdict
ImportError: No module named _collections