FPGAwars/apio #148: apio on raspberry pi zero with icezero

rdwrt, 2018-03-01:
Hi! 

I am trying out apio to see if it works on the raspberry pi zero with the icezero

https://blackmesalabs.wordpress.com/2017/02/07/icezero-fpga-board-for-rasppi/

https://shop.trenz-electronic.de/en/TE0876-02-Ice-Zero-with-Lattice-ICE-ICE40HX

The icezero uses a ICE40HX4K which should be similar to some of the already supported boards.

I seem to get into an issue with the raspberry pi zero, as it runs on linux_armv6l and so far it seems only packages exist for linux_armv7l 

(apio) pi@raspberrypi:~ $ apio install --all
Installing examples package:
Download apio-examples-0.0.10.zip
Downloading  [####################################]  100%             
Unpacking  [####################################]  100%             
Package 'examples' has been successfully installed!
Installing icestorm package:
Warning: full platform does not match: linux_armv6l                
         Trying OS name: linux                
Error: Got an unrecognized status code '404' when downloaded https://github.com/FPGAwars/toolchain-icestorm/releases/download/v1.10.1/toolchain-icestorm-linux-1.10.1.tar.gz

I guess it should be possible to create linux_armv6l versions of the packages that already exist as linux_armv7l ?

rdwrt, 2018-03-01:
I think I should take this to https://github.com/FPGAwars/toolchain-icestorm first

Jesus89, 2018-03-03:
Hi @rdwrt.

The main packages for apio are not available for `linux_armv6l` yet:
- [Toolchain-icestorm](https://github.com/FPGAwars/toolchain-icestorm)
- [Toolchain-iverilog](https://github.com/FPGAwars/toolchain-iverilog)
- [Tools-system](https://github.com/FPGAwars/tools-system)

We need to build all these packages for `linux_armv6l` in order to have support for this architecture in apio.

I created several branches to start with it but I couldn't test it since I haven't got an RPi Zero. Also it seems that it is not trivial to build source code for `linux_armv6l`. If you want to try the builds, here you have the branches:
- [Toolchain-icestorm linux-armv6l branch](https://github.com/FPGAwars/toolchain-icestorm/tree/linux-armv6l)
- [Toolchain-iverilog linux-armv6l branch](https://github.com/FPGAwars/toolchain-iverilog/tree/linux-armv6l)
- [Tools-system linux-armv6l branch](https://github.com/FPGAwars/tools-system/tree/linux-armv6l)

If you have good results in cross-compiling these packages let me know in this ticket or with a PR contribution :)

Thanks!

rdwrt, 2018-03-04:
Ok, thanks, very useful pointers, I will start experimenting!

lagomorph, 2021-08-21:
I assume no progress was made on this?

zapta, 2025-05-07:
The list of supported platforms was updated in the apio dev branch to the list below. If there is any pending program with the dev branch, please file a new issue with updated details.

```
$ apio info platforms

Apio Supported Platforms                                     
┌────────────────────────┬──────────────────────────────────┐
│ PLATFORM ID            │ DESCRIPTION                      │
├────────────────────────┼──────────────────────────────────┤
│   darwin               │ Mac OSX, x86 64 bit              │
├────────────────────────┼──────────────────────────────────┤
│ * darwin_arm64         │ Mac OSX, ARM 64 bit              │
├────────────────────────┼──────────────────────────────────┤
│   linux_aarch64        │ Linux ARM 64 bit                 │
├────────────────────────┼──────────────────────────────────┤
│   linux_x86_64         │ Linux X86 64 bit                 │
├────────────────────────┼──────────────────────────────────┤
│   windows_amd64        │ Windows x86 64 bit               │
└────────────────────────┴──────────────────────────────────┘
```

Meanwhile, closing this one.