FPGAwars/apio #158: apio build throws exception for fpga101

jelly, 2018-06-02:
With the latest release and master from a virtualenv `apio build` throws the following exception

```shell
[jelle@lithium][~/projects/fpga101-workshop/tests/led]%apio build
Board: fpga101
Error: 'NoneType' object has no attribute 'get'
```

This all due to the boards json missing the fpga101 board. Error handling should however be a lot better and indicate what the issue is.

jelly, 2018-06-02:
Note that the just released beta works, but if any board in apio.ini does not exist in the json file you'll get this obscure error.

Jesus89, 2018-06-03:
Hi. You are right, this is not a nice error. I noticed that this only happens when the board is configured in the `apio.ini` file and not when it is passed as an argument:

```
$ apio build -b X
Info: ignore apio.ini board
Error: unknown board: X
```

Let's show the same error for this case.

Jesus89, 2018-06-03:
Closed by aff504f95e8169f65155c87e0562aaabec3e4cd4

closed 2018-06-03 by Jesus89