FPGAwars/apio #159: Error: the JSON object must be str, not 'bytes'

gudenau, 2018-07-16:
I followed the dirrections here to try and program a TinyFPGA BX that I just got and when I get to the upload step it errors with `Error: the JSON object must be str, not 'bytes'`. I guess this is some random Python issue?

What happened:
Uploading the example project errors out with `Error: the JSON object must be str, not 'bytes'`

What should happen:
The project should upload and the FPGA should run it.

Platform:
`uname -a` : `Linux DESKTOP-DEBIAN 4.9.0-6-amd64 #1 SMP Debian 4.9.88-1+deb9u1 (2018-05-07) x86_64 GNU/Linux`

Steps to reproduce: (or what I did)
`sudo apt-get install python3-pip`
`sudo pip3 install --upgrade --no-cache-dir apio==0.4.0b3 "tinyprog>=1.0.9"`
`apio install system scons icestorm iverilog` (since drivers does not exist)
`apio drivers --serial-enable`
`tinyprog --update-bootloader`
Download atom-amd64.deb
`sudo dpkg -i atom-amd64.deb`
Install `apio-ide` in atom.
`git clone https://github.com/tinyfpga/TinyFPGA-BX.git`
`cp TinyFPGA-BX/apio_template blink_project`
Open `blink_project` with `atom`.
Upload.
Error happens here.

Issue originally reported to TinyFPGA, found to be an APIO issue so I am riposting it here.
tinyfpga/TinyFPGA-BX#2

Jesus89, 2018-07-20:
Hi @gudenau.

The command `apio upload` calls `tinyprog -m` to get metadata in order to verify the TinyFPGA BX device before uploading the bitstream.

This is basically the code for that:

```python
result = exec_command(['tinyprog', '-m'])
    if result:
        return json.loads(result.get('out'))
```

So I think the issue is that `tinyprog -m` output is not always in JSON format. Could you paste the output of several  `tinyprog -m` after a successfully `apio build`?

Jesus89, 2018-07-21:
cc @tinyfpga. Could you take a look at this issue adding a print here?

```python
result = exec_command(['tinyprog', '-m'])
    if result:
        print(result)
        return json.loads(result.get('out'))
```

Is possible that `tinyprog -m` does not generate proper output sometimes?

gudenau, 2018-07-21:
Sorry, was busy with other stuff.

```
gudenau@DESKTOP-DEBIAN:~$ tinyprog -m
[
  {
    "boardmeta": {
      "name": "TinyFPGA BX",
      "fpga": "ice40lp8k-cm81",
      "uuid": "XXXXXXXX-XXXX-XXXX-XXXX-XXXXXXXXXXXX",
      "hver": "1.0.0"
    },
    "bootmeta": {
      "bver": "1.0.1",
      "addrmap": {
        "userdata": "0x50000-0x100000",
        "userimage": "0x28000-0x50000",
        "bootloader": "0x000a0-0x28000"
      },
      "update": "https://tinyfpga.com/update/tinyfpga-bx",
      "bootloader": "TinyFPGA USB Bootloader"
    },
    "port": "/dev/ttyACM0"
  }
]
```

Please note, the UUID was a proper UUID with [0-9a-f], I just like to remove that sort of thing.

ewenmcneill, 2018-07-22:
In case it helps, `exec_command` is using `subprocess.Popen()` and `.communicate()` on that, and the [Python 3 `subprocess.Popen.communicate` man page](https://docs.python.org/3/library/subprocess.html#subprocess.Popen.communicate) notes that `out` will be a `str` if the streams were opened in `text` mode, and `bytes` otherwise.  [`exec_command`](https://github.com/FPGAwars/apio/blob/aff504f95e8169f65155c87e0562aaabec3e4cd4/apio/util.py#L307-L339) appears to leave the stream mode (text/binary) up to the caller, and [`get_tinyprog_meta`](https://github.com/FPGAwars/apio/blob/aff504f95e8169f65155c87e0562aaabec3e4cd4/apio/util.py#L441-L446) does not do anything to make it text (so my guess is at least on Linux/Unix Python will default to binary/bytes).  The [`Popen` constructor](https://docs.python.org/3/library/subprocess.html#popen-constructor) takes a `text=True` argument, which might be sufficient to fix this issue.

[Before Python 3.6, `json.loads()` expects to receive a `str` *not* `bytes`/`bytearray`](https://docs.python.org/3/library/json.html#basic-usage) -- from Python 3.6 onwards, it will also accept `bytes`/`bytearray`.  So this appears to be an "older Python 3.x" compatibility issue -- Python 2.7, and Python 3.6+ seem likely to work.  Debian Stretch and Ubuntu 16.04 both use Python 3.5, which *is* affected by this.  (Eg, [Debian Stretch python3 package](https://packages.debian.org/stretch/python3).)

That perhaps explains why relatively few users are affected -- Python 3.6 has enough backwards compatibility updates to make it work, and the TinyFPGA BX example is Python 3.6....

In theory `text=True` passed through the chain to `Popen` would be sufficient to make it work everywhere for this specific use case

Ewen

DurandA, 2018-07-22:
I got the same problem. Unfortunately, it cannot be fixed using `text=True` as it is not available in Python3.5 [Popen constructor](https://docs.python.org/3.5/library/subprocess.html#popen-constructor).

mharsch, 2018-07-30:
setting:

`subprocess.Popen(universal_newlines=True)`

fixes this issue (for the tinyprog -m case).

tinyfpga, 2018-07-30:
I had a similar bug with `tinyprog` running on certain installs of Python 3.5.  `tinyprog` would crash with this same error when parsing the JSON metadata stored in the TinyFPGA BX SPI flash.  I'll have to go back to my commits and see what the fix was.  I have found that the `six` module from PYPI is very useful in taking care of compatibility issues across multiple versions of 2.x and 3.x

Jesus89, 2018-08-15:
Hi. Thanks to all for the information. I have verified that the bug is only in Python 3.5. It can be solved by using `.decode('utf-8')` and this change does not break Python 2.7 or Python 3.6 :)

```python
json.loads(result.get('out').decode('utf-8'))
```

ewenmcneill, 2018-08-17:
@Jesus89 `apio` 0.3.4 appears to work with Python 3.5 on Ubuntu 16.04 LTS.  Thanks!

Of note, I found I had to uninstall/reinstall `apio` (using `pip`) because it appears I had previously installed  0.4.0b3 ([following the TinyFPGA BX Guide](https://tinyfpga.com/bx/guide.html)), and after doing that I also had to do `apio install icestorm` because `apio` 0.3.4 wanted an earlier `icestorm` package to use (1.10.1 not 1.11.0-beta, which presumably `apio`0.4.0b3 had installed).  Just a heads up if anyone else wants to switch to the fixed 0.3.4 version for this TinyFPGA BX issue, that you might need to uninstall/reinstall `apio` and `icestorm` to do so.

Thanks for applying the fix!

Ewen

PS: Possibly the same fix could be made to the 0.4 beta branch too?

PPS: Yes, all that progress output did display on multiple lines, and apparently repeatably so (rather than just updating the same console line).  Unsure why.

    (tinyfpga) ewen@parthenon:/src/tinyfpga/examples/blink$ python --version
    Python 3.5.2
    (tinyfpga) ewen@parthenon:/src/tinyfpga/examples/blink$ pip list 2>&1 | egrep 'apio|tinyprog'
    apio             0.3.4    
    tinyprog         1.0.18   
    (tinyfpga) ewen@parthenon:/src/tinyfpga/examples/blink$ tinyprog -m
    [
      {
        "bootmeta": {
          "update": "https://tinyfpga.com/update/tinyfpga-bx",
          "addrmap": {
            "userdata": "0x50000-0x100000",
            "userimage": "0x28000-0x50000",
            "bootloader": "0x000a0-0x28000"
          },
          "bver": "1.0.1",
          "bootloader": "TinyFPGA USB Bootloader"
        },
        "port": "/dev/ttyACM0",
        "boardmeta": {
          "name": "TinyFPGA BX",
          "fpga": "ice40lp8k-cm81",
          "uuid": "d02fde2d-xxxx-xxxx-xxxx-xxxxxxc6f3d1",
          "hver": "1.0.0"
        }
      }
    ]
    (tinyfpga) ewen@parthenon:/src/tinyfpga/examples/blink$ apio upload
    Warning: `icestorm` package version 1.11.0-beta
    does not match the semantic version >=1.10.1,<1.11.0
    Please run:
       apio install icestorm
    (tinyfpga) ewen@parthenon:/src/tinyfpga/examples/blink$ apio install icestorm
    Installing icestorm package:
    Download toolchain-icestorm-linux_x86_64-1.10.1.tar.gz
    Downloading  [####################################]  100%             
    Unpacking  [####################################]  100%
    Package 'icestorm' has been successfully installed!
    (tinyfpga) ewen@parthenon:/src/tinyfpga/examples/blink$ apio upload
    [Sat Aug 18 10:51:53 2018] Processing TinyFPGA-BX
    --------------------------------------------------------------------------------
    yosys -p "synth_ice40 -blif hardware.blif" -q top.v
    arachne-pnr -d 8k -P cm81 -p pins.pcf -o hardware.asc -q hardware.blif
    [.... build warnings omitted to save space ...]
    icepack hardware.asc hardware.bin
    tinyprog --pyserial -c /dev/ttyACM0 --program hardware.bin
    
    Erasing:   0%|          | 0.00/135k [00:00<?, ?B/s]
    Erasing:  24%|██▍       | 32.8k/135k [00:00<00:00, 145kB/s]
    Erasing:  73%|███████▎  | 98.3k/135k [00:00<00:00, 149kB/s]
    Erasing:  97%|█████████▋| 131k/135k [00:00<00:00, 154kB/s]
    Erasing: 100%|██████████| 135k/135k [00:00<00:00, 156kB/s]
    
    Writing:   0%|          | 0.00/135k [00:00<?, ?B/s]
    Writing:  14%|█▍        | 19.2k/135k [00:00<00:00, 191kB/s]
    Writing:  28%|██▊       | 38.1k/135k [00:00<00:00, 191kB/s]
    Writing:  42%|████▏     | 57.1k/135k [00:00<00:00, 190kB/s]
    Writing:  57%|█████▋    | 77.3k/135k [00:00<00:00, 193kB/s]
    Writing:  72%|███████▏  | 97.0k/135k [00:00<00:00, 193kB/s]
    Writing:  85%|████████▌ | 115k/135k [00:00<00:00, 189kB/s]
    Writing:  98%|█████████▊| 133k/135k [00:00<00:00, 186kB/s]
    Writing: 100%|██████████| 135k/135k [00:00<00:00, 188kB/s]
    
    Reading:   0%|          | 0.00/135k [00:00<?, ?B/s]
    Reading:  35%|███▍      | 46.9k/135k [00:00<00:00, 469kB/s]
    
    TinyProg CLI
    ------------
    Using device id 1d50:6130
    Programming /dev/ttyACM0 with hardware.bin
    Programming at addr 028000
    Waking up SPI flash
    135100 bytes to program
    Success!
    
    Reading:  69%|██████▉   | 93.6k/135k [00:00<00:00, 467kB/s]
    Reading: 100%|██████████| 135k/135k [00:00<00:00, 466kB/s]
    ========================= [SUCCESS] Took 4.02 seconds =========================
    (tinyfpga) ewen@parthenon:/src/tinyfpga/examples/blink$

Jesus89, 2018-08-18:
Hello. Thanks for the tests and the detailed information.

Apio 0.3.4 uses the stable toolchain (https://github.com/FPGAwars/toolchain-icestorm/releases/tag/v1.10.1) and Apio 0.4.0-beta uses the latest toolchain. That's why version 0.3.4 requires to update the toolchain.

I released apio 0.3.4 with the stable toolchain and TinyFPGA BX support because it is required for the Icestudio release, but of course I'll apply the fix in the 0.4 beta release too.

The next step to do in the 0.4 version is to create an stable release of the toolchain with the latest changes (icestorm, yosys, arachne-pnr, iverilog, ...).

Regarding the progressbar I saw the same issue. With Python 2 the char `#` is shown in the progressbar, and in Python 3 there is the problem you're referring to. This is because `tinyprog` uses tqdm and when I execute `tinyprog` from a subprocess it does not behave the same. I'll investigate more on that.

ewenmcneill, 2018-08-18:
Thanks for the detailed explanation.  It's nice that there's a release version (0.3.4) with TinyFPGA BX support.  Possibly @tinyfpga might want to now update the [TinyFPGA BX User Guide](https://tinyfpga.com/bx/guide.html) to suggest people install `apio` 0.3.4 rather than the 0.4.0-beta (which I assume was required before), since `apio` 0.3.4 should now be the most compatible version.

Other than that, I suspect it's probably safe to close this ticket now.  The `subprocess` handling of the programming status output could probably be looked at separately, as it's only a minor visual issue (and I'd guess something is behaving differently depending on whether it thinks it is outputting to a terminal or not).

Ewen

jwrussert, 2019-03-09:
For the record, I had this same error while following the instructions on tinyfpga.com. They have a "Tips" section at the end of the walkthrough that suggests how to update the bootloader. I just used these two commands and was able to update the bootloader and move on.

`pip install --upgrade --no-cache-dir tinyprog
tinyprog --update-bootloader`

TuxRobotics, 2020-03-13:
I was still running into this issue.  

Changing line 128 in `/usr/local/lib/python3.5/dist-packages/tinyprog/__main__.py`

From
`bootloader_update_info = json.loads(urlopen(m[u"bootmeta"][u"update"] + "/bootloader.json").read())`

To

`bootloader_update_info = json.loads(urlopen(m[u"bootmeta"][u"update"] + "/bootloader.json").read().decode("utf-8"))`

Solved the issue for me.

closed 2020-04-09 by Obijuan