FPGAwars/apio #160: -D VCD_OUTPUT unused?

x8-999-github, 2018-07-26:
When running simlulation in apio a parameter is given to iverilog. This parameter is later used to determine the name of the output file. On my setup this does not work (and the replacement does not happen with the effect that my test files become  `VCD_OUTPUT.vcd (including the `)

```
iverilog -B "/home/why/.apio/packages/toolchain-iverilog/lib/ivl" -o leds_tb.out -D VCD_OUTPUT=leds_tb "/home/why/.apio/packages/toolchain-iverilog/vlib/cells_sim.v" leds.v leds_tb.v
vvp -M "/home/why/.apio/packages/toolchain-iverilog/lib/ivl" leds_tb.out
VCD info: dumpfile `VCD_OUTPUT.vcd opened for output.
End of simulation
gtkwave leds_tb.vcd leds_tb.gtkw
Gtk-Message: 12:53:44.533: Failed to load module "canberra-gtk-module"
 
￼
GTKWave Analyzer v3.3.86 (w)1999-2017 BSI
 
Error opening  .vcd file 'leds_tb.vcd'.
Why: No such file or directory
scons: *** [sim] Error 255
========================= [ ERROR ] Took 0.61 seconds =========================
```

[image]

[image]

Jesus89, 2018-08-18:
Hi. I have released apio 0.3.4, that uses the stable toolchain. Could you install this version and test it again?

The stable toolchain has Icarus Verilog v10_1, and the latest toolchain (use in apio 0.4.0 beta) the version v10_2. Maybe there is a different in the management of this flag between versions.

x8-999-github, 2018-08-30:
Indeed "downgrading" to 0.3.4 solved my problem ! thanks

closed 2018-08-30 by x8-999-github