FPGAwars/apio #161: vvp readline dependency

x8-999-github, 2018-07-26:
On ubuntu bionic libreadline 6 is replace by libreadline 7. hence when vvp gets executing it can not find the library

The quite ugly hack I did was to symlink the libraries 
``
why@login:~/.apio/packages/toolchain-iverilog/bin$ dpkg -S /usr/lib/x86_64-linux-gnu/libhistory.so 
libreadline-dev:amd64: /usr/lib/x86_64-linux-gnu/libhistory.so
why@login:~/.apio/packages/toolchain-iverilog/bin$ apt-cache policy libreadline-dev
libreadline-dev:
  Installed: 7.0-3
  Candidate: 7.0-3
  Version table:
 *** 7.0-3 500
        500 http://nl.archive.ubuntu.com/ubuntu bionic/main amd64 Packages
        100 /var/lib/dpkg/status
why@login:~/.apio/packages/toolchain-iverilog/bin$ ./vvp 
./vvp: error while loading shared libraries: libhistory.so.6: cannot open shared object file: No such file or directory
why@login:~/.apio/packages/toolchain-iverilog/bin$ ./vvp 
./vvp: error while loading shared libraries: libreadline.so.6: cannot open shared object file: No such file or directory
```

After creating the symlinks:
```
why@login:~/.apio/packages/toolchain-iverilog/bin$ ./vvp 
./vvp: no input file.
```

Jesus89, 2018-08-18:
Thanks for the information. The `vvp` package is the only one that is not statically build. I'll try to build it more statically binary for the next release.

Jesus89, 2018-08-22:
Already published `toolchain-iverilog` packages with all binaries static:

- v1.1.1 (apio 0.3)
- v1.2.1 (apio 0.4)

Just run the following command to update the package:

```
apio install iverilog@1.1.1
```

or

```
apio install iverilog@1.2.1
```

closed 2018-08-22 by Jesus89