FPGAwars/apio #162: apio build - parameter for top-level module?

gundy, 2018-08-02:
Hi,

Something I noticed today is that `apio build` doesn't seem to have a parameter to pass in a top-level module to use..

I've got a project which is split over a few files, and ran into issues this morning where seemingly the last module passed in on the yosys command line (in my case that was done alphabetically) is assumed to be the top-level module.

It's entirely possible that I'm missing something, but I have tried checking the docs and couldn't find anything there.

Thanks,
Dave.

Jesus89, 2018-08-15:
Hi. There is no explicit option (yet) for this feature but you can use a workaround:

Execute `apio init --scons` to create a `SConstruct` file in your project's directory. You can modify this script to your needs and apio will use this one instead of the default SConstruct.

In your case simply edit [this line](https://github.com/FPGAwars/apio/blob/develop/apio/resources/SConstruct#L172) to add the yosys command:

```
yosys -p \"synth_ice40 -top mytop -blif $TARGET\" {} $SOURCES
```

I'll add in the future the top module feature.

gundy, 2018-08-17:
Great! Thanks! I'll close this off now - the workaround will be fine 👍

closed 2018-08-17 by gundy