FPGAwars/apio #163: apio install client forbidden https

nottledim, 2018-08-02:
I'm trying to run "apio install system" but it complains: Error: 403 Client Error: Forbidden for url: https://api.github.com/repos/FPGAwars/tools-system/releases

I can download that URL with browser.  I assume this is a certificate verification issue but it's not clear to me what's wrong.

Could somebody help me debug this please.

Debian GNU/Linux "stretch"

nottledim, 2018-08-15:
On 08/15/18 02:19, Jesús Arroyo Torrens wrote:
> Hi. Is this happening when you install other packages?

all the apio install <xyz> attempts fail with the same error.

> Have you tried different networks? 

as in different home locations?  No.  What I have done is install on a
different computer when this problem did not occur.  That suggests the problem
is local to my specific workstation.

What version of Python are you using?

2.7 and 3.5 have been tried.

Using python interactively:

Python 3.5.3 (default, Jan 19 2017, 14:11:04)
[GCC 6.3.0 20170118] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> import requests
>>> requests.get('https://github.com')
<Response [200]>
>>> requests.get('https://api.github.com')
<Response [403]>

So it's something to do with api.github.com

Any ideas?

Dick

Jesus89, 2018-08-15:
I tried the same tests:

```
Python 3.5.2 (default, Nov 23 2017, 16:37:01) 
[GCC 5.4.0 20160609] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> import requests
>>> requests.get('https://github.com')
<Response [200]>
>>> requests.get('https://api.github.com')
<Response [200]>
```

Maybe you can execute `requests.get('https://api.github.com').json()` to get more information about the error in your machine.

nottledim, 2018-08-17:
On 08/15/18 20:46, Jesús Arroyo Torrens wrote:

Resolved.  It took a while to track down.  It was due to a 'default' entry in
~/.netrc which *overrides* the 'Authorization' header in the request.

It seems to me that that is a defect in python-requests but I've read
discussions where that is debated
(https://github.com/requests/requests/issues/3929).

> Maybe you can execute |requests.get('https://api.github.com').json()| to get
> more information about the error in your machine.

The symptom was http error 403 "Maximum number of login attempts exceeded.
Please try again later".  github issue that error and message for
authentication failures.

FYI the headers can be seen with:

import requests
r=requests.get('https://api.github.com')
r.request.headers

Thanks for your help

Jesus89, 2018-08-17:
Great. Congrats!

closed 2018-08-17 by Jesus89

simonjwright, 2018-12-06:
@nottledim Thanks for this! I found ~/.netrc from 2005 (three ISPs ago). Delete; all is well.