FPGAwars/apio #164: Supported boards in v0.3.3 vs documentation

zakqwy, 2018-08-02:
I am trying to use apio with a TinyFPGA BX and while this board appears to be supported by the project (see README.md or apio/resources/boards.json), the latest installed version does not seem to include this device or several others from the docs:
apio --version
`apio, version 0.3.3`
apio boards --list
`Supported boards:

--------------------------------------------------------------------------------
Board         FPGA                 Type  Size  Pack      
--------------------------------------------------------------------------------
Cat-board     iCE40-HX8K-CT256     hx    8k    ct256     
TinyFPGA-B2   iCE40-LP8K-CM81      lp    8k    cm81      
alhambra-ii   iCE40-HX4K-TQ144     hx    8k    tq144:4k  
blackice      iCE40-HX4K-TQ144     hx    8k    tq144:4k  
blackice-ii   iCE40-HX4K-TQ144     hx    8k    tq144:4k  
go-board      iCE40-HX1K-VQ100     hx    1k    vq100     
iCE40-HX8K    iCE40-HX8K-CT256     hx    8k    ct256     
icestick      iCE40-HX1K-TQ144     hx    1k    tq144     
icezum        iCE40-HX1K-TQ144     hx    1k    tq144     
icoboard      iCE40-HX8K-CT256     hx    8k    ct256     
kefir         iCE40-HX4K-TQ144     hx    8k    tq144:4k `

I believe this is causing the following error:
apio upload
`Board: TinyFPGA-BX
Error: 'NoneType' object has no attribute 'get'`

tinyfpga, 2018-08-02:
The TinyFPGA BX is supported in a beta release.  To install a beta release from pypi you can specify the version:

pip install --upgrade --no-cache-dir apio==0.4.0b3

zakqwy, 2018-08-02:
Thanks Luke, this allowed me to build the project with apio.

closed 2018-08-02 by zakqwy