FPGAwars/apio #165: Add support for testing

RobertBaruch, 2018-08-05:
It would be nice to have an "apio test" command. The tests could be organized like [TinyFPGA's tests here](https://github.com/tinyfpga/TinyFPGA-Bootloader/tree/master/tests).

```
<project dir>/
  apio.ini
  file1.v
  file2.v
  tests/
    utility_header1.vh
    utility_header2.vh
    dothing_test/
      test.v
    dootherthing_test/
      test.v
    another_test/
      test.v
```

So for each tests/*_test/test.v file, the build command would be:

```
iverilog -Itests -s top_tb -o tests/TARGET_DIR/test <all files in project dir> tests/TARGETDIR/test.v
tests/TARGET_DIR/test
```

Hopefully the `.vcd` file output from the test will also end up in `tests/TARGET_DIR`, otherwise cd to that directory first, then run the test.

Then, all I have to do is run `apio test` and it will go through all the tests, run them, and I'll have unit-tested my project.

zapta, 2024-11-12:
The 'apio test' command is implemented and will be included in next apio release. It finds and runs all the files that match *_tb.v.   

Help text
```
$ apio test -h
Usage: apio test [OPTIONS] [TESTBENCH_FILE]

  The sim command simulates one or all the testbenches in the project and is
  useful for automatic unit testing of the code. Testbenches are expected to
  exist with the $fatal directive if any error is detected. The commands is
  typically used in the root directory of the project that contains the
  apio.ini.

  Examples
    apio test                 # Run all *_tb.v testbenches.
    apio test my_module_tb.v  # Run a single testbench

  For a sample testbench that is compatible with apio see the example at
  https://github.com/FPGAwars/apio-examples/tree/master/upduino31/testbench

  [Hint] To simulate the testbench with a graphical visualizaiton of the
  signals see the apio sim command.

Options:
  -p, --project-dir path  Set the root directory for the project.
  -h, --help              Show this message and exit.
```

Sample log
```
 $ apio test
Loading project's custom 'boards.json' file.
Setting the envinronment.
iverilog -B "/Users/user/.apio/packages/tools-oss-cad-suite/lib/ivl" -o ftdi_tx_tb.out -DVCD_OUTPUT=ftdi_tx_tb -DNO_ICE40_DEFAULT_ASSIGNMENTS "/Users/user/.apio/packages/tools-oss-cad-suite/share/yosys/ice40/cells_sim.v" ftdi_tx.v i2s_rx.v i2s_test_pattern.v i2s_timing.v leds.v main.v queue.v queue_pusher.v register.v reset_gen.v ftdi_tx_tb.v
vvp -M "/Users/user/.apio/packages/tools-oss-cad-suite/lib/ivl" ftdi_tx_tb.out
VCD info: dumpfile ftdi_tx_tb.vcd opened for output.
End of simulation
ftdi_tx_tb.v:201: $finish called at 7700000 (1ps)
iverilog -B "/Users/user/.apio/packages/tools-oss-cad-suite/lib/ivl" -o i2s_rx_tb.out -DVCD_OUTPUT=i2s_rx_tb -DNO_ICE40_DEFAULT_ASSIGNMENTS "/Users/user/.apio/packages/tools-oss-cad-suite/share/yosys/ice40/cells_sim.v" ftdi_tx.v i2s_rx.v i2s_test_pattern.v i2s_timing.v leds.v main.v queue.v queue_pusher.v register.v reset_gen.v i2s_rx_tb.v
vvp -M "/Users/user/.apio/packages/tools-oss-cad-suite/lib/ivl" i2s_rx_tb.out
VCD info: dumpfile i2s_rx_tb.vcd opened for output.
End of simulation
i2s_rx_tb.v:132: $finish called at 13300000 (1ps)
iverilog -B "/Users/user/.apio/packages/tools-oss-cad-suite/lib/ivl" -o i2s_timing_tb.out -DVCD_OUTPUT=i2s_timing_tb -DNO_ICE40_DEFAULT_ASSIGNMENTS "/Users/user/.apio/packages/tools-oss-cad-suite/share/yosys/ice40/cells_sim.v" ftdi_tx.v i2s_rx.v i2s_test_pattern.v i2s_timing.v leds.v main.v queue.v queue_pusher.v register.v reset_gen.v i2s_timing_tb.v
vvp -M "/Users/user/.apio/packages/tools-oss-cad-suite/lib/ivl" i2s_timing_tb.out
VCD info: dumpfile i2s_timing_tb.vcd opened for output.
End of simulation
i2s_timing_tb.v:48: $finish called at 141100000 (1ps)
iverilog -B "/Users/user/.apio/packages/tools-oss-cad-suite/lib/ivl" -o leds_tb.out -DVCD_OUTPUT=leds_tb -DNO_ICE40_DEFAULT_ASSIGNMENTS "/Users/user/.apio/packages/tools-oss-cad-suite/share/yosys/ice40/cells_sim.v" ftdi_tx.v i2s_rx.v i2s_test_pattern.v i2s_timing.v leds.v main.v queue.v queue_pusher.v register.v reset_gen.v leds_tb.v
vvp -M "/Users/user/.apio/packages/tools-oss-cad-suite/lib/ivl" leds_tb.out
VCD info: dumpfile leds_tb.vcd opened for output.
End of simulation
leds_tb.v:83: $finish called at 9710000 (1ps)
iverilog -B "/Users/user/.apio/packages/tools-oss-cad-suite/lib/ivl" -o main_tb.out -DVCD_OUTPUT=main_tb -DNO_ICE40_DEFAULT_ASSIGNMENTS "/Users/user/.apio/packages/tools-oss-cad-suite/share/yosys/ice40/cells_sim.v" ftdi_tx.v i2s_rx.v i2s_test_pattern.v i2s_timing.v leds.v main.v queue.v queue_pusher.v register.v reset_gen.v main_tb.v
vvp -M "/Users/user/.apio/packages/tools-oss-cad-suite/lib/ivl" main_tb.out
VCD info: dumpfile main_tb.vcd opened for output.
End of simulation
main_tb.v:105: $finish called at 15770000 (1ps)
iverilog -B "/Users/user/.apio/packages/tools-oss-cad-suite/lib/ivl" -o queue_pusher_tb.out -DVCD_OUTPUT=queue_pusher_tb -DNO_ICE40_DEFAULT_ASSIGNMENTS "/Users/user/.apio/packages/tools-oss-cad-suite/share/yosys/ice40/cells_sim.v" ftdi_tx.v i2s_rx.v i2s_test_pattern.v i2s_timing.v leds.v main.v queue.v queue_pusher.v register.v reset_gen.v queue_pusher_tb.v
vvp -M "/Users/user/.apio/packages/tools-oss-cad-suite/lib/ivl" queue_pusher_tb.out
VCD info: dumpfile queue_pusher_tb.vcd opened for output.
End of simulation
queue_pusher_tb.v:269: $finish called at 6300000 (1ps)
iverilog -B "/Users/user/.apio/packages/tools-oss-cad-suite/lib/ivl" -o queue_tb.out -DVCD_OUTPUT=queue_tb -DNO_ICE40_DEFAULT_ASSIGNMENTS "/Users/user/.apio/packages/tools-oss-cad-suite/share/yosys/ice40/cells_sim.v" ftdi_tx.v i2s_rx.v i2s_test_pattern.v i2s_timing.v leds.v main.v queue.v queue_pusher.v register.v reset_gen.v queue_tb.v
vvp -M "/Users/user/.apio/packages/tools-oss-cad-suite/lib/ivl" queue_tb.out
VCD info: dumpfile queue_tb.vcd opened for output.
End of simulation
queue_tb.v:90: $finish called at 5900000 (1ps)
iverilog -B "/Users/user/.apio/packages/tools-oss-cad-suite/lib/ivl" -o register_tb.out -DVCD_OUTPUT=register_tb -DNO_ICE40_DEFAULT_ASSIGNMENTS "/Users/user/.apio/packages/tools-oss-cad-suite/share/yosys/ice40/cells_sim.v" ftdi_tx.v i2s_rx.v i2s_test_pattern.v i2s_timing.v leds.v main.v queue.v queue_pusher.v register.v reset_gen.v register_tb.v
vvp -M "/Users/user/.apio/packages/tools-oss-cad-suite/lib/ivl" register_tb.out
VCD info: dumpfile register_tb.vcd opened for output.
End of simulation
[... cut ...] x.v i2s_test_pattern.v i2s_timing.v leds.v main.v queue.v queue_pusher.v register.v reset_gen.v reset_gen_tb.v
vvp -M "/Users/user/.apio/packages/tools-oss-cad-suite/lib/ivl" reset_gen_tb.out
VCD info: dumpfile reset_gen_tb.vcd opened for output.
End of simulation
reset_gen_tb.v:38: $finish called at 5300000 (1ps)
======================================================== [SUCCESS] Took 0.56 seconds ========================================================
/Users/user/projects/fpga/repo/hdl $ 

```

zapta, 2024-11-12:
Implemented. Closing.

closed 2024-11-12 by zapta