FPGAwars/apio #168: [Apio 0.3.4] Error: Invalid semantic version (>=1.1.0,<1.2.0)

Obijuan, 2018-08-23:
OS: Ubuntu 18.04
Fresh installation of apio with **sudo pip install apio**

```
$ apio --version
apio, version 0.3.4
```

When I install all the packages, I get the error
```
$ apio install --all
Installing examples package:
Download apio-examples-0.0.11.zip
Downloading  [####################################]  100%             
Unpacking  [####################################]  100%
Package 'examples' has been successfully installed!
Installing icestorm package:
Download toolchain-icestorm-linux_x86_64-1.10.1.tar.gz
Downloading  [####################################]  100%             
Unpacking  [####################################]  100%
Package 'icestorm' has been successfully installed!
Error: Invalid semantic version (>=1.1.0,<1.2.0)
```

The problems seem to be with the **iverilog** package:

```
$ apio install iverilog
Error: Invalid semantic version (>=1.1.0,<1.2.0)
```

In this animation the process to reproduce the error are shown:

[image]

Jesus89, 2018-08-24:
Apio `0.3.5` released with the install command fixed https://github.com/FPGAwars/apio/commit/a67dfaa59c0803e932260a69248b4ee96f716a95.

Obijuan, 2018-08-24:
Thanks! I've tested on Ubuntu 18.04. It works! :-). 
I'll close the issue

closed 2018-08-24 by Obijuan