FPGAwars/apio #173: External Toolchain is not recognized

martin2250, 2019-02-24:
Both icestorm and scons are installed on my system (Archlinux), but apio still prompts me to install them using the internal package manager:
```
$which icepack
/usr/bin/icepack
$which scons
/usr/bin/scons
$apio build
Error: scons toolchain is not installed
Please run:
   apio install scons
Error: icestorm toolchain is not installed
Please run:
   apio install icestorm
```
Does apio require specific versions of these packages? If not, apio should check for the availability of the needed commands instead of checking against the internal package manager.

Jesus89, 2019-09-08:
Yes, It requires specific versions of the packages. In this case:
- https://github.com/FPGAwars/tool-scons
- https://github.com/FPGAwars/toolchain-icestorm

However, I'll revisit it to make it more flexible in case you have the packages already installed in your system.

Obijuan, 2022-05-01:
Scons is no longer an internal apio package. It is installed as an apio dependecy

closed 2022-05-01 by Obijuan