FPGAwars/apio #175: SB_HFOSC usage

AloyseTech, 2019-02-24:
Hi,

Experimenting with the framework, a simple blink using the SB_HFOSC module would not verify using `appio verify`.

blink.v
```verilog
module blinky (output wire BLUn,
               output wire GRNn,
               output wire REDn);

   wire clk;
   SB_HFOSC inthosc(.CLKHFPU(1'b1), .CLKHFEN(1'b1), .CLKHF(clk));

   localparam N = 27;
   reg [N:0] counter;

   always @(posedge clk)
     counter <= counter + 1;

   assign led_blue = counter[N];
   assign led_green = counter[N-1];
   assign led_red = counter[N-2];
endmodule
```

Error : 
```
iverilog -o hardware.out -D VCD_OUTPUT= "C:\users\USERNAME\.apio\packages\toolchain-iverilog\vlib\cells_sim.v" blinky.v
blinky.v:6: error: Unknown module type: SB_HFOSC
2 error(s) during elaboration.
*** These modules were missing:
SB_HFOSC referenced 1 times.
***
scons: *** [hardware.out] Error 2
```

s-ol, 2019-03-09:
quick and dirt fix is adding

```
module SB_HFOSC(
    input CLKHFPU,
    input CLKHFEN,
    output CLKHF
);
parameter CLKHF_DIV = "0b00";
endmodule
```

at the end of `.apio\packages\toolchain-iverilog\vlib\cells_sim.v`.

s-ol, 2019-03-09:
The problem is actually not that this definition is missing, it is provided in `apio/packages/toolchain-icestorm/share/yosys/ice40/cells_sim.v`, but `apio verify' (wrongly?) includes `apio/packages/toolchain-iverilog/vlib/cells_sim.v`, which doesn't include the more specific modules.

`apio build` does read the correct library though.

koppanyh, 2019-03-31:
I also had this error.

I found that copying the 'iverilog -o ...' part of the error, modifying the path to use the icestorm toolchain, and running it with the apio raw command worked for me.

I'm using Ubuntu 18.04, so mine looked something like this:
`apio raw 'iverilog -B "/home/USERNAME/.apio/packages/toolchain-iverilog/lib/ivl" -o hardware.out -D VCD_OUTPUT= "/home/USERNAME/.apio/packages/toolchain-icestorm/share/yosys/ice40/cells_sim.v" blinky.v'`

The error I got with apio verify was this:
`USERNAME@HOSTNAME:~/Desktop/upduino$ apio verify
iverilog -B "/home/USERNAME/.apio/packages/toolchain-iverilog/lib/ivl" -o hardware.out -D VCD_OUTPUT= "/home/USERNAME/.apio/packages/toolchain-iverilog/vlib/cells_sim.v" blinky.v
blinky.v:6: error: Unknown module type: SB_HFOSC
2 error(s) during elaboration.
*** These modules were missing:
SB_HFOSC referenced 1 times.
***
scons: *** [hardware.out] Error 2
========================= [ ERROR ] Took 0.23 seconds =========================`

Obijuan, 2019-12-23:
It should now be fixed in the apio/develop branch. It would be great if you could check whether it is working on not. 

I will close this issue. Feel free to reopen it again if it does not work properly

closed 2019-12-23 by Obijuan