FPGAwars/apio #189: dyld: Library not loaded: /usr/local/opt/libftdi/lib/libftdi1.2.dylib

bayoussef, 2019-11-20:
Hi,

Using the blink.v program for my Upduino 2, attempting to upload the bitstream to the board using "apio upload" on mac terminal. I get the error :

dyld: Library not loaded: /usr/local/opt/libftdi/lib/libftdi1.2.dylib
Referenced from: /Users/baba/.apio/packages/tools-system/bin/lsftdi
Reason: image not found

trylaarsdam, 2021-12-08:
Were you able to solve this issue? Having the same error when trying `apio system --lsftdi`

Obijuan, 2021-12-08:
I will keep issue open until more people can test it  (When we finish the release of icestudio 0.8 we will start a new cycle and we will continue with the process of updating the toolchains/solving issues)

trylaarsdam, 2021-12-08:
[image]
To add to this, the error also occurs in the latest version of icestudio on an M1 macbook running big sur.

Obijuan, 2022-05-01:
This error should have been fixed in apio 0.8.1. Please try it and feel free to re-open this issue if it remains (or open a new issue if a new bug is found). Thanks for the feedback

closed 2022-05-01 by Obijuan