FPGAwars/apio #191: Python error when enabling serial I/O, ref. #147

lbattraw, 2019-12-11:
When running to enable serial I/O it produces a type error:

apio drivers --serial-enable
Configure Serial drivers for FPGA
Traceback (most recent call last):
  File "/home/lcb/.local/bin/apio", line 10, in <module>
    sys.exit(cli())
  File "/home/lcb/.local/lib/python3.7/site-packages/click/core.py", line 722, in __call__
    return self.main(*args, **kwargs)
  File "/home/lcb/.local/lib/python3.7/site-packages/click/core.py", line 697, in main
    rv = self.invoke(ctx)
  File "/home/lcb/.local/lib/python3.7/site-packages/click/core.py", line 1066, in invoke
    return _process_result(sub_ctx.command.invoke(sub_ctx))
  File "/home/lcb/.local/lib/python3.7/site-packages/click/core.py", line 895, in invoke
    return ctx.invoke(self.callback, **ctx.params)
  File "/home/lcb/.local/lib/python3.7/site-packages/click/core.py", line 535, in invoke
    return callback(*args, **kwargs)
  File "/home/lcb/.local/lib/python3.7/site-packages/click/decorators.py", line 17, in new_func
    return f(get_current_context(), *args, **kwargs)
  File "/home/lcb/.local/lib/python3.7/site-packages/apio/commands/drivers.py", line 32, in cli
    exit_code = Drivers().serial_enable()
  File "/home/lcb/.local/lib/python3.7/site-packages/apio/managers/drivers.py", line 99, in serial_enable
    return self._serial_enable_linux()
  File "/home/lcb/.local/lib/python3.7/site-packages/apio/managers/drivers.py", line 165, in _serial_enable_linux
    group_added = self._add_dialout_group()
  File "/home/lcb/.local/lib/python3.7/site-packages/apio/managers/drivers.py", line 196, in _add_dialout_group
    if 'dialout' not in groups:
TypeError: a bytes-like object is required, not 'str'**

The problem is that the object being passed is a byte array type and needs to be decoded before being compared.  I stuck a print statement on the line before the error, and this is the data in the object:

b'lcb adm dialout cdrom sudo audio dip plugdev lpadmin sambashare docker vboxsf\n'

The offending line is:

groups = subprocess.check_output('groups')

Adding a decode fixes the error, though it could be a problem if the data is passed as a string instead of byte type at some time in the future.

groups = subprocess.check_output('groups').decode()

Obijuan, 2019-12-23:
It should be fixed now in the apio/develop branch
It would be great if you could test it again
Clone the develop branch, enter the apio folder and install  apio-develop executing the following command:

sudo python3 setup.py install

Then try the apio drivers --serial-enable  and check if it returns without errors

Obijuan, 2019-12-23:
I will close this issue, as it is working ok. If you find problems, feel free to reopen this issue

closed 2019-12-23 by Obijuan