FPGAwars/apio #195: apio, version 0.5.3: scons: *** Do not know how to make...

UweBonnes, 2020-02-24:
Hello,

some hours ago. I installed apio and could get some simple NCO design to work. Great tool!
However when playing with the options, I hit "apio upgrade" and now have 0.5.3. 

In the directory, where building succeeded before, I now get:

apio build
[Mon Feb 24 19:12:39 2020] Processing icestick
--------------------------------------------------------------------------------------------------------------------------------------------------------------------------------

scons: warning: Ignoring missing SConscript '/home/bon/.local/lib/python3.6/site-packages/apio/resources/ice40/SConstruct'
File "/home/bon/.apio/packages/tool-scons/script/scons", line 201, in <module>
scons: *** Do not know how to make File target `build' (/devel/apio/nco/build).  Stop.

Do I need to clean up or update more things?

UweBonnes, 2020-02-24:
This happens too when installing apio on another machine...

UweBonnes, 2020-02-24:
Cloning apio and "cp -pr ice40 /home/bon/.local/lib/python3.6/site-packages/apio/resources/" fixed the problem. So the distrubution file is missing files!

MikeyBoo, 2020-02-25:
Having the same issue. Decided to just revert to an older version but lost support of Fomu in doing so.

cavearr, 2020-02-26:
Hi! we are working on it! i'm updating the progress in this issue.

cavearr, 2020-02-26:
I think this is fixed! can you try it?

pip install -U apio  

Thanks!

UweBonnes, 2020-02-26:
Thanks, I removed .local/lib/python3.6/site-packages/apio/resources/ice40, did "pip install -U apio" and ic40 and ecp5 got recreated and "apio build" works

closed 2020-02-26 by UweBonnes