FPGAwars/apio #198: Add support for the Orange Crab

Obijuan, 2020-04-09:
For supporting the Orange Crab board in icestudio (see [this issue in Icestudio](https://github.com/FPGAwars/icestudio/issues/392)) it should be first supported in Apio

Obijuan, 2020-04-09:
The first step is to add the board information in this file: [boards.json](https://github.com/FPGAwars/apio/blob/master/apio/resources/boards.json)
@jmi2k could you help on this?

jmi2k, 2020-04-09:
I'm on it, but right now the VID/PID pair used on the DFU bootloader conflicts with Fomu. The author has [requested a new PID here](https://github.com/pidcodes/pidcodes.github.com/pull/499) which I'm going to use temporarily as a placeholder.

jmi2k, 2020-04-23:
I've updated the PR info, as he finally got the PID. Now it can be merged.

Obijuan, 2020-04-24:
Great! Thanks a lot! 🙂

Obijuan, 2020-04-25:
I am reopening this issue again. The building and bitstream generation works fine, but not the upload. It need more testing and polishing, as reported by @gregdavill in this icestudio's issue [#392](https://github.com/FPGAwars/icestudio/issues/392)

jmi2k, 2020-04-25:
The issue he is raising seems unrelated to this PR specifically. He has more than one DFU programmer attached to his computer, so the flashing step is complaining because it cannot choose one by default. Can anyone test flashing something to a DFU-based board while having another DFU-based board connected at the same time?

Obijuan, 2020-04-26:
It is also important to remove the sudo command as it is not available in other platforms (like Windows). I will open another issue for that

The solution is to add a udev rule when executing an apio drivers command (that should be defined, it can be called dfu-enable)

PR #204 (2020-05-04): Refactor previous attempts to add udev rules for DFU devices
I've removed the auto-generation and installation of udev rules, now these are hardcoded into apio/resources/80-fpga-ftdi.rules with the rest of the rules.

This closes #198 and #201 (as this should be enough to make dfu-util available to non-root users).
closed 2020-05-04 by Obijuan