FPGAwars/apio #201: new command for configurgin dfu as a driver

Obijuan, 2020-04-26:
Currently, for programing boards that uses the dfu-util command what is used is calling sudo dfu-util.  The problem is that it is not multi-platform and it cannot be used inside icestudio (as you have to type the password in a console that is not shown)

jmi2k, 2020-04-26:
AFAIK other projects handle this by creating a group for themselves, giving it all the rights they need. Then, users only have to add them to that group. This sadly is a manual step which the user has to do, but as long as it is well documented it shouldn't be a problem.

Then, adding an udev rule should be as easy as getting a list of all the devices which use `dfu-util`, and generating the appropiate rules.

jmi2k, 2020-04-26:
I've submitted a draft PR (#202), it can be tested. Just create the `apio` group in your linux system, add yourself to that group and run setup.py as root. Then plug a DFU device and check the group of the device in `/dev/bus/<bus>/<device>`, it should be `apio`!

Obijuan, 2020-04-26:
We can use that by now for testing,  but then it should be integrated into the [drivers.py](https://github.com/FPGAwars/apio/blob/master/apio/managers/drivers.py) file, where all that kind of stuff is done

closed 2020-05-04 by Obijuan