FPGAwars/apio #205: Toolchain: Support for iceprogduino

Obijuan, 2020-05-04:
The OLIMEX iCE40HX1K-EVB board is programed with "iceprogduino" which can be found in
https://github.com/OLIMEX/iCE40HX1K-EVB/tree/master/programmer

Iceprogduino is a c-program which comes with a Makefile ("make + sudo make install")

In order to fully support the Olimex board in Apio this programmer should be included as part of the apio packages

x653, 2021-02-13:
Hi Obijuan,
I send a merge request to toolchain-icestorm in which i added the script compile_iceprogduino.sh.
Does this help?

Obijuan, 2021-02-18:
for sure! It will help a lot :-)  Thanks!

xralphack, 2021-06-23:
@x653 
iceprogduino hardware.bin
sh: iceprogduino: command not found
scons: *** [upload] Error 127

Do you know how to fix this?

xralphack, 2021-06-23:
My temporary way:

Download the https://github.com/OLIMEX/iCE40HX1K-EVB/tree/master/programmer repo
Update the SerialPort in source code
Compile
cp iceprogduino /usr/local/bin/iceprogduino

then I can use apio upload

x653, 2021-06-23:
iceprogduino is a c-program used to upload the bitstream file to Olimexino fpga board iCE40HX1/8K-EVB connected over an arduino like programmer board (e.g. olimexino-32u4)

To install iceprogduino:
1. clone the repository
git clone https://github.com/OLIMEX/iCE40HX1K-EVB.git

2. cd into the direcory
cd iCE40HX1K-EVB/programmer/iceprogduino

3. compile and install
make
sudo make install

4. Run iceprogduino
iceprogduino

Hope this helps.

By the way: To use iceprogduino you have to install the firmware onto your programmer board (olimexino-32u4). The firmware can be found in the same repository. Upload with arduino IDE...

xralphack, 2021-06-23:
@x653 Thanks for your reply.
I used the same way as you mentioned. It works now.

Obijuan, 2024-03-26:
iceprogduino is already been included in apio

closed 2024-03-26 by Obijuan