FPGAwars/apio #208: Packaged version of nextpnr-ice40 doesn't support --gui option

mcknly, 2020-05-08:
Trying to launch the layout viewer (see [nextpnr](https://github.com/YosysHQ/nextpnr) readme) and it seems the apio packaged version of nextpnr-ice40 does not support --gui.

`~/.apio/packages/toolchain-ice40/bin/nextpnr-ice40 --json hardware.json --pcf pins.pcf --asc hardware.asc --gui`
`unrecognised option '--gui'`

I am not sure when the GUI feature was added, is the build just to far behind or is there some issue with dependencies?

mmicko, 2020-05-08:
Apio includes static linked nextpnr and that is without gui and python. This makes linux package portable and usable on multiple different distributions.

mcknly, 2020-05-08:
Ah okay. So I will need to build myself to leverage that functionality. I will close the issue, thanks

closed 2020-05-08 by mcknly