FPGAwars/apio #210: PLL on iCEstick not generated correctly

damianolodi, 2020-08-26:
Hi everyone,
    I am using APIO to compile a working code that was written by a colleague and previously compiled using the *iCEcube2* program. The board on which it is uploaded is the iCEstick evaluation kit.

In this program we need to use a PLL at 240MHz, and the compilation using the iCEcube2 software works correctly. Unfortunately, when building with APIO, I think the PLL is not working correctly (and in fact the FPGA behave differently based on which program I load).

Using the command `apio build -v` I obtain the following outputs (I report only the output that seems relevant to me. I the all output is needed, I will upload it):
```
Info: Placing PLLs..
Info:   constrained PLL 'fastClk.mother_pll_inst' to X6/Y0/pll_3
Info: Packing special functions..
Warning: PLL 'fastClk.mother_pll_inst' has unknown unconnected port 'DYNAMICDELAY' - ignoring
...
Info: Device utilisation:
Info: 	         ICESTORM_LC:   525/ 1280    41%
Info: 	        ICESTORM_RAM:     0/   16     0%
Info: 	               SB_IO:     7/  112     6%
Info: 	               SB_GB:     8/    8   100%
Info: 	        ICESTORM_PLL:     1/    1   100%
Info: 	         SB_WARMBOOT:     0/    1     0%
...
Info: Max frequency for clock 'fclk_$glb_clk': 95.74 MHz (PASS at 12.00 MHz)

Info: Max delay <async>               -> posedge fclk_$glb_clk: 8.18 ns
Info: Max delay posedge fclk_$glb_clk -> <async>              : 1.13 ns
...
Info: Critical path report for clock 'fclk_$glb_clk' (posedge -> posedge):
...
Info: Max frequency for clock 'fclk_$glb_clk': 98.89 MHz (PASS at 12.00 MHz)
...
Info: Max delay <async>               -> posedge fclk_$glb_clk: 6.54 ns
Info: Max delay posedge fclk_$glb_clk -> <async>              : 1.13 ns
```

I am not very expert in FPGA and Verilog, so I may be wrong, but from the outputs it seems to me that the PLL was created but the output frequency was not the required one (240 MHz). The requested code parameters are correct according to the `icepll` tool.

In the following I will report the code used to generate the PLL.

**File `mother_pll.v`**
```verilog
module mother_pll(REFERENCECLK,
                  PLLOUTCORE,
                  PLLOUTGLOBAL,
                  RESET);

input REFERENCECLK;
input RESET;    /* To initialize the simulation properly, the RESET signal (Active Low) must be asserted at the beginning of the simulation */ 
output PLLOUTCORE;
output PLLOUTGLOBAL;

SB_PLL40_CORE mother_pll_inst(.REFERENCECLK(REFERENCECLK),
                              .PLLOUTCORE(PLLOUTCORE),
                              .PLLOUTGLOBAL(PLLOUTGLOBAL),
                              .EXTFEEDBACK(),
                              .DYNAMICDELAY(),
                              .RESETB(RESET),
                              .BYPASS(1'b0),
                              .LATCHINPUTVALUE(),
                              .LOCK(),
                              .SDI(),
                              .SDO(),
                              .SCLK());

//\\ Fin=12, Fout=240;
defparam mother_pll_inst.DIVR = 4'b0000;
defparam mother_pll_inst.DIVF = 7'b1001111;
defparam mother_pll_inst.DIVQ = 3'b010;
defparam mother_pll_inst.FILTER_RANGE = 3'b001;
defparam mother_pll_inst.FEEDBACK_PATH = "SIMPLE";
defparam mother_pll_inst.DELAY_ADJUSTMENT_MODE_FEEDBACK = "FIXED";
defparam mother_pll_inst.FDA_FEEDBACK = 4'b0000;
defparam mother_pll_inst.DELAY_ADJUSTMENT_MODE_RELATIVE = "FIXED";
defparam mother_pll_inst.FDA_RELATIVE = 4'b0000;
defparam mother_pll_inst.SHIFTREG_DIV_MODE = 2'b01;
defparam mother_pll_inst.PLLOUT_SELECT = "GENCLK";
defparam mother_pll_inst.ENABLE_ICEGATE = 1'b0;

endmodule
```

**Main file `mother.v`.**
```verilog
mother_pll fastClk (
    .REFERENCECLK(clk),
    .PLLOUTCORE(fclk),
    .PLLOUTGLOBAL(PLLOUTGLOBAL),
    .RESET(1'b1)
    );
```

**`clk` constrain in `pcf` file**
```
set_io clk  21
```

Had you some similar problem?
Thank you in advance for your help.

hstarmans, 2020-09-17:
As far as I understand,  your circuit can only work with clocks up to 98.89 MHz or 95.74 MHz. It doesn't say what PLL clock was generated but if you use this clock for logic it will not work. Typically the requirements are passed to nextpnr via the prepack argument but apio is out of date so you can't pass these constraints. You could try to use yowasp-nextpnr-ice40.

zapta, 2025-05-07:
As @hstarmans  explained, the toolchain reported the max frequency of this design as 95.74 MHz and the OP interpreted it as the actual PLL frequency. 

We can close this one. Please reopen if the issue requires further discussion.

closed 2025-05-07 by zapta