FPGAwars/apio #211: Perform a new release

pcossutta, 2020-09-02:
This is a request, not an issue. Next month we are starting to use upduino3 in an online University course. It would be of great help to have the board supported in the apio release. As the board is currently supported on GitHub a new release would allow us to use apio without any modification. Is that possible ?

By the way I think that the usage of apio.ini should be different. If the board data where on it instead of only the label of the platform, one could provide a custom apio.ini to support a new board without the need to release a new version. I think that would also improve the usability of apio as well. I couldnt find an easy way to do this without modifying boards.json and given that every user possible have a different os or different python distribution, this is a difficult task to perform.

Obijuan, 2020-11-22:
Ups! I forgot to leave a comment here. I am so sorry for the late response  
It is not necesary to have a release for testing the latest changes in apio, just install the development version with this command:

```
sudo pip3 install -U git+https://github.com/FPGAwars/apio.git@develop#egg=apio
```

So, whenever someone make a change, they emits a pull-req and it is available after the merge, intallling apio with the previous command

Nevertheless, we will try to release apio as soon as possible. We still have to finish the support for the ulx and fomu boards

More information on apio develpment version can be found here:

https://github.com/FPGAwars/apio/wiki/Apio-development-version

pcossutta, 2020-11-24:
Thank you !!! I will close this as this isn't an issue anymore.

closed 2020-11-24 by pcossutta

Obijuan, 2020-11-25:
At last we have a new release: apio 0.6.0:  https://pypi.org/project/apio/0.6.0/
Release cycles will be shorter from now on