FPGAwars/apio #212: Integrate the binaries from the FPGA-toolchain

Obijuan, 2020-10-03:
Currently the toolchain used in apio is compiled for the different platformts using isolated building scripts located in different repos. The FPGA-toolchain project has already integrated almost everything into one repo were the nightly builds are generated. It would be great if apio could get the binaries from there

Obijuan, 2020-11-25:
The dfu-utils has been already integrated from [FPGA-toolchain](https://github.com/open-tool-forge/fpga-toolchain) to the [toolchain-dfu apio package](https://github.com/FPGAwars/toolchain-dfu)

Roadmap:

- [x] dfu-utils
- [ ] Yosys
- [ ] ECP5
- [ ] ICE40

nobodywasishere, 2021-03-19:
[I tried getting](https://github.com/nobodywasishere/toolchain-yosys) the `yosys` binary to work by pulling the scripts from `toolchain-dfu` and modifying them for yosys, but run into errors with using yosys. It may have been just due to some nightly glitch though.
The error from running yosys: `ERROR: init_share_dirname: unable to determine share/ directory!`. Not sure how to fix this as I've copied all the yosys related stuff. 

Also, the directory.json in apio will have to be updated to remove the upper limit on the version, had to build apio from scratch to get around this restriction.

Obijuan, 2022-05-01:
The oss-cad-suite is used for the apio toolchain. The migration process has been finish in apio 0.8.1

closed 2022-05-01 by Obijuan