FPGAwars/apio #218: APIO missed changes in memory initialization files and does not build new binaries

EvgenyMuryshkin, 2020-10-30:
Hi,

I found that only changes to Verilog source cause build to run again. Changes in memory init files are ignored.

Here is the test repo

https://github.com/EvgenyMuryshkin/qusoc/tree/apio-missed-changes/qusoc/tinyfpga

1. Clone the repo
2. Checkout "apio-missed-changes" branch
3. Open "qusoc/tinyfpga" folder in Atom with APIO
4. Build
5. Change the content of SimpleCounterModule_TopLevel_State_buff.hex file
6. Build again

The second build does not do anything but expected to run build again with a modified memory file

Thanks,
Regards,
Evgeny

zapta, 2024-11-12:
This is a scons dependency issue.  A verilog includes an hex file but apio doesn't detect that dependency.

https://github.com/EvgenyMuryshkin/qusoc/blob/4277b7c4bcc50826ba102f27b4f4f848619438f7/qusoc/tinyfpga/SimpleCounterModule_TopLevel.v#L56

We can resolve this in a few ways. For example

* Automatic detection, using a dependency detector similar to the one we use for icestudio  https://github.com/FPGAwars/apio/blob/303dbddce93daf9eb0c7794a4a9391d6bae7e7fc/apio/scons/scons_util.py#L309

* Semi automatic, allowing the user to insert dependency markers within their files, e.g. 
```
// apio-dependency: "my_other_file.hex"
```

* Manual, adding in apio.ini specification of files that all the .v files depends on.

```
dependencies:  my_file1.hex, my_file2.data
```

zapta, 2024-11-12:
Once apio.ini will be required, we will make it also a dependency for all other files.

zapta, 2024-11-26:
The file reference is here

https://github.com/EvgenyMuryshkin/qusoc/blob/4277b7c4bcc50826ba102f27b4f4f848619438f7/qusoc/tinyfpga/SimpleCounterModule_TopLevel.v#L56

zapta, 2024-12-23:
Detection on $readmemh reference implemented. Closing.