FPGAwars/apio #219: ICEsugar board support

Obijuan, 2020-11-10:
Give support to the [icesugar](https://github.com/wuxx/icesugar/) board  
This support was started by @aalku   
In this issue we are documenting the state and the tests  

Roadmap:

- [ ] Verilog blinky example in apio-example. For testing purposes it is a good idea to add a blinky example that can be download directly from apio using the apio example -d command
- [ ] icesprog toolchain for the platforms: Linux, mac and windows. This is the repo for storing the binaries: [toolchain-icesprog](https://github.com/FPGAwars/toolchain-icesprog)
- [ ] Testing the icesugar. Make sure that apio upload works ok in at least in two of the three platforms: Linux, Mac and windows

Obijuan, 2020-11-11:
Blinky example ready, added to the [apio-examples packages](https://github.com/FPGAwars/apio-examples)  
@aalku, it would be nice if you could test it. I will post the instruction on the FPGAwars forum

aalku, 2020-11-12:
I ran the build commands successfully, programmed it and it worked flaulessly. :)

These are the commands and results:

```
C:\Users\UserName\Documents\GitHub\apio-upstream\x>apio examples -d iCESugar_1_5/Blinky
Creating iCESugar_1_5/Blinky directory ...
Example 'iCESugar_1_5/Blinky' has been successfully created!

C:\Users\UserName\Documents\GitHub\apio-upstream\x>cd iCESugar_1_5/Blinky

C:\Users\UserName\Documents\GitHub\apio-upstream\x\iCESugar_1_5\Blinky>apio build
[Thu Nov 12 19:06:45 2020] Processing iCESugar_1_5
-------------------------------------------------------------------------------------------------------------------------------------
yosys -p "synth_ice40 -json hardware.json" -q blink.v
nextpnr-ice40 --up5k --package sg48 --json hardware.json --asc hardware.asc --pcf icesugar.pcf -q
Warning: unmatched constraint 'SW[0]' (on line 7)
Warning: unmatched constraint 'SW[1]' (on line 8)
Warning: unmatched constraint 'SW[2]' (on line 9)
Warning: unmatched constraint 'SW[3]' (on line 10)
Warning: unmatched constraint 'RX' (on line 14)
Warning: unmatched constraint 'TX' (on line 15)
Warning: unmatched constraint 'USB_DP' (on line 17)
Warning: unmatched constraint 'USB_DN' (on line 18)
Warning: unmatched constraint 'USB_PULLUP' (on line 19)
Warning: unmatched constraint 'P1_1' (on line 22)
Warning: unmatched constraint 'P1_2' (on line 23)
Warning: unmatched constraint 'P1_3' (on line 24)
Warning: unmatched constraint 'P1_4' (on line 25)
Warning: unmatched constraint 'P1_9' (on line 26)
Warning: unmatched constraint 'P1_10' (on line 27)
Warning: unmatched constraint 'P1_11' (on line 28)
Warning: unmatched constraint 'P1_12' (on line 29)
Warning: unmatched constraint 'P2_1' (on line 32)
Warning: unmatched constraint 'P2_2' (on line 33)
Warning: unmatched constraint 'P2_3' (on line 34)
Warning: unmatched constraint 'P2_4' (on line 35)
Warning: unmatched constraint 'P2_9' (on line 36)
Warning: unmatched constraint 'P2_10' (on line 37)
Warning: unmatched constraint 'P2_11' (on line 38)
Warning: unmatched constraint 'P2_12' (on line 39)
Warning: unmatched constraint 'P3_1' (on line 42)
Warning: unmatched constraint 'P3_2' (on line 43)
Warning: unmatched constraint 'P3_3' (on line 44)
Warning: unmatched constraint 'P3_4' (on line 45)
Warning: unmatched constraint 'P3_9' (on line 46)
Warning: unmatched constraint 'P3_10' (on line 47)
Warning: unmatched constraint 'P3_11' (on line 48)
Warning: unmatched constraint 'P3_12' (on line 49)
Warning: unmatched constraint 'P4_1' (on line 52)
Warning: unmatched constraint 'P4_2' (on line 53)
Warning: unmatched constraint 'P4_3' (on line 54)
Warning: unmatched constraint 'P4_4' (on line 55)
Warning: unmatched constraint 'SPI_SS' (on line 58)
Warning: unmatched constraint 'SPI_SCK' (on line 59)
Warning: unmatched constraint 'SPI_MOSI' (on line 60)
Warning: unmatched constraint 'SPI_MISO' (on line 61)
41 warnings, 0 errors
icepack hardware.asc hardware.bin
C:\Users\UserName\.apio\packages\tool-scons\script\..\engine\SCons\Defaults.py:213: SyntaxWarning: "is not" with a literal. Did you mean "!="?
if len(operation_list) is not 2:
==================================================== [SUCCESS] Took 3.62 seconds ====================================================

C:\Users\UserName\Documents\GitHub\apio-upstream\x\iCESugar_1_5\Blinky>

C:\Users\UserName\Documents\GitHub\apio-upstream\x\iCESugar_1_5\Blinky>c:\Users\UserName\.icestudio\apio\packages\toolchain-icesprog\bin\icesprog.exe hardware.bin
flash offset: 0x00000000
write flash (104090 (0x1969a) Bytes)
write 0x00000000
write 0x00010000
done

C:\Users\UserName\Documents\GitHub\apio-upstream\x\iCESugar_1_5\Blinky>
```

And you can see here the blinky effect:

https://youtu.be/ahHLP2JzkJw

https://youtu.be/wEDMbXawgDk

Obijuan, 2020-11-13:
Thanks a lot for the tests!!!!   I've sent you a new battery of test, in the [FPGAwars group](https://groups.google.com/g/fpga-wars-explorando-el-lado-libre/c/Gg4oYklfH4k). I've already upload the icesprog executables for Widowns and Linux. Now it should be possible to install it with apio install icesprog both in Linux and Windows.

aalku, 2020-11-13:
As I said in the FPGAwars group (in spanish) on Windows it runs flaulessly.
Unfortunately on Linux it seems to work only as root.

You have the commands results in my answer [here](https://groups.google.com/g/fpga-wars-explorando-el-lado-libre/c/Gg4oYklfH4k/m/YLp0CuFeAgAJ)

aalku, 2020-11-13:
It fails here. Run as root is in the left:

[image]

I run hidtest and you can see the difference there too. It can enumerate the devices but as non privileged user it can't show device details:

```
ubuntu@ubuntu:~/apio_test/iCESugar_1_5/Blinky$ /home/ubuntu/.apio/packages/toolchain-icesprog/bin/hidtest 
Device Found
  type: 1d50 602b
  path: 0002:0005:03
  serial_number: (null)
  Manufacturer: (null)
  Product:      (null)
  Release:      100
  Interface:    3

Device Found
  type: 0e0f 0003
  path: 0002:0002:00
  serial_number: (null)
  Manufacturer: (null)
  Product:      (null)
  Release:      103
  Interface:    0

unable to open device
ubuntu@ubuntu:~/apio_test/iCESugar_1_5/Blinky$ sudo /home/ubuntu/.apio/packages/toolchain-icesprog/bin/hidtest 
Device Found
  type: 1d50 602b
  path: 0002:0005:03
  serial_number: 07000001066cff343832434257103306a5a5a5a597969908
  Manufacturer: MuseLab
  Product:      DAPLink CMSIS-DAP
  Release:      100
  Interface:    3

Device Found
  type: 0e0f 0003
  path: 0002:0002:00
  serial_number: (null)
  Manufacturer: VMware
  Product:      VMware Virtual USB Mouse
  Release:      103
  Interface:    0

unable to open device
ubuntu@ubuntu:~/apio_test/iCESugar_1_5/Blinky$ 

```

I'll read about hidapi and root issues.

aalku, 2020-11-13:
Ok. I think I undestand. I think we need to add udev stuff here:
https://github.com/FPGAwars/apio/blob/5bcf7913b2987ff744d8bafa8454dd5c156c6c20/apio/managers/drivers.py#L61

Now I'm sure you knew this. :)

aalku, 2020-11-13:
After adding this to `/etc/udev/rules.d/80-fpga-ftdi.rules`:
```
# -- for iCESugar board
ATTRS{idVendor}=="1d50", ATTRS{idProduct}=="602b", MODE="0660", GROUP="plugdev", TAG+="uaccess"
```

and reconnecting the board it works: 

```
ubuntu@ubuntu:~/apio_test/iCESugar_1_5/Blinky$ apio upload
[Fri Nov 13 08:42:57 2020] Processing iCESugar_1_5
-----------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------
icesprog hardware.bin
flash offset: 0x00000000
write flash (104090 (0x1969a) Bytes)
write 0x00000000
write 0x00010000
done
======================================================================================= [SUCCESS] Took 9.71 seconds =======================================================================================
ubuntu@ubuntu:~/apio_test/iCESugar_1_5/Blinky$ 
```

😍

I don't think `hid` and `ftdi` are the same so maybe it should be in another file.

It seems hid stands for Human Interface Device (keyboard, mouse...) while FTDI is actually an enterprise related to USB-Serial devices. I didn't know it was an enterprise. I thought it was a "tecnology name".

Well, I'm sure you'll know what to do with this. I learned about USB devices permissions. :)

Obijuan, 2020-11-14:
Great! Suport for the icesugar in windows: Finished! 

Support for the icesugar in Linux:
I've already include the new udev rules in apio. Just for the safe side, I would like you to test it again. The new tests are proposed on the FPGAwars group

aalku, 2020-11-14:
Done!

```
		ubuntu@ubuntu:~/apio-test/iCESugar_1_5/Blinky$ apio drivers --ftdi-disable
		Revert FTDI drivers configuration
		FTDI drivers disabled
		Unplug and reconnect your board
		ubuntu@ubuntu:~/apio-test/iCESugar_1_5/Blinky$ apio upload
		[Sat Nov 14 09:57:54 2020] Processing iCESugar_1_5
		--------------------------------------------------------------------------------
		icesprog hardware.bin
		iCELink open fail!
		scons: *** [upload] Error 255
		========================= [ ERROR ] Took 0.20 seconds =========================
		ubuntu@ubuntu:~/apio-test/iCESugar_1_5/Blinky$ apio drivers --ftdi-enable
		Configure FTDI drivers for FPGA
		FTDI drivers enabled
		Unplug and reconnect your board
		ubuntu@ubuntu:~/apio-test/iCESugar_1_5/Blinky$ apio upload
		[Sat Nov 14 09:58:05 2020] Processing iCESugar_1_5
		--------------------------------------------------------------------------------
		icesprog hardware.bin
		flash offset: 0x00000000
		write flash (104090 (0x1969a) Bytes)
		write 0x00000000
		write 0x00010000
		done
		========================= [SUCCESS] Took 9.50 seconds =========================
```

Obijuan, 2020-11-15:
great! Thanks a lot! I am updating the roadmap. Now the only item left is the support for mac. I need @cavearr to give me a hand on that. I need to connect to a mac virtual machine to perform the tests and to generate the executable

Obijuan, 2020-11-22:
The toolchain is ready for mac!!! :-)  
For testing in mac we need both a mac and the icesugar. That is complicated. So, the final step: upload has not yet been tested, but I will close this issue

If a mac user with an icesugar board test it and it fails, please reopen this issue

closed 2020-11-22 by Obijuan

Archfx, 2023-02-16:
Hi,

I am a Mac user with an IceSugar board. Apparently, I cant install the icesprog with `apio install icesprog` command.

Following is the error message
```shell
❯ apio install icesprog
(DEBUG) Profile path: /Users/archfx/.apio/profile.json
(DEBUG) Home_dir: /Users/archfx/.apio
Error: no such package 'icesprog'
```
Am I doing something wrong here?

Any help would be appreciated. Thanks

Obijuan, 2023-02-16:
The package icesprog is no longer available. Instead you should install oss-cad-suite