FPGAwars/apio #220: Support for VHDL: Use yosys built together with the ghdl-yosys-plugin, common SConstruct

bl0x, 2020-11-27:
I've tried to extend the apio SConstruct to also support a VHDL toolchain, making use of yosys and the ghdl-yosys-plugin for synthesis. For simulation, plain ghdl is used instead.

The code is [here](https://github.com/bl0x/fstr_project). It includes example code in vhdl and verilog and a common SConstruct file.

It would be nice, if apio would also include ghdl and the ghdl plugin for yosys, perhaps from the [fpga-toolchain] (https://github.com/open-tool-forge/fpga-toolchain) repo.

What do you think? Is this a viable path?

Obijuan, 2020-11-27:
Yes! I think it is viable. Thank you a lot! Using the binaries from FPGA-toolchain is proposed on issue #212  
The steps would be:
- [ ] Creation of an apio package for yosys from the fpga-toolchain
- [ ] Including you example in the apio-examples packages. This makes it very easy for anyone to test it. This is something that you can do. [The instrucctions are here](https://github.com/FPGAwars/apio/wiki/The-apio-examples-package)
- [ ] Perform tests and debug the external SConstruct
- [ ] Integrate the new SConstruct in apio

As there are two SConstruct files currently, for the ice40 and ECP5 families, maybe it would be also a good idea to merge them into one SConstruct capable of managing everything: verilog and vhdl for both ECP5 and ICE40 (and future families)

bl0x, 2020-11-27:
I've made a pull request to the apio-examples repo.

manuelpas, 2020-12-15:
https://github.com/FPGAwars/icestudio/issues/382

PPlinux, 2021-05-26:
Dear,

is vhdl entry via ghdl yosys plugin already supported apio ?

Greetings,

Patrick Pelgrims

bl0x, 2021-05-28:
Dear @PPlinux , yes. At least somewhat. The example mentioned here uses a special SConstruct file that also has support for VHDL input. I've been using this now on a couple of projects, and it works quite well. You can also mix VHDL and verilog code in the same project.

PPlinux, 2021-05-28:
Dear,

Where can I find such a construct file ?

Greetings,
Patrick

bl0x, 2021-05-28:
Check this out:

https://github.com/FPGAwars/apio-examples/blob/master/TinyFPGA-BX/clock_divider/vhdl/SConstruct

manuelpas, 2021-05-29:
how install ghdl-yosys-plugin with apio?

PPlinux, 2021-05-30:
> Check this out:
> 
> https://github.com/FPGAwars/apio-examples/blob/master/TinyFPGA-BX/clock_divider/vhdl/SConstruct

And how should we use this construct (file) ?
Use it like make file or configuration file  used by GHDL etc ?

PPlinux, 2021-05-30:
> how install ghdl-yosys-plugin with apio?

A good link to how-to-install GHDL-Yosys-Nextpnr ... On Windows/Linux and some nice examples would be a nice help.

bl0x, 2021-06-01:
> 
> 
> > Check this out:
> > https://github.com/FPGAwars/apio-examples/blob/master/TinyFPGA-BX/clock_divider/vhdl/SConstruct
> 
> And how should we use this construct (file) ?
> Use it like make file or configuration file used by GHDL etc ?

apio uses the SConstruct file by default, if it exists in the project directory.
```
apio build
```

mjcherbert, 2021-09-18:
I'm quite interested in using this mechanism, I did manage to build yosys and the ghdl plugin in a stand-alone manner, would like to get this working with apio ... not so fussed if icestudio doesn't integrate with it at the moment

sebeaumont, 2022-03-25:
> Yes! I think it is viable. Thank you a lot! Using the binaries from FPGA-toolchain is proposed on issue #212 The steps would be:
> 
> * [ ]  Creation of an apio package for yosys from the fpga-toolchain
> * [x]  Including your example in the apio-examples packages. This makes it very easy for anyone to test it. This is something that you can do. [The instrucctions are here](https://github.com/FPGAwars/apio/wiki/The-apio-examples-package)
> * [ ]  Perform tests and debug the external SConstruct
> * [ ]  Integrate the new SConstruct in apio
> 
> As there are two SConstruct files currently, for the ice40 and ECP5 families, maybe it would be also a good idea to merge them into one SConstruct capable of managing everything: verilog and vhdl for both ECP5 and ICE40 (and future families)

Is there any WIP on this? -- it's also worth noting that if I do `apio init -s` in an ECP5 project I get the wrong one (it's looking for .pcf file)   -- maybe kill a couple of birds with one stone and have one `SConstruct` to rule them all? I have had good success with GHDL plugin -- be great to support both out of the box.

closed 2025-05-07 by zapta