FPGAwars/apio #222: Confusion about versions - where can 6.0.0 be found?

achary, 2020-12-13:
I've installed `apio` using PIP and I have version `0.6.0` installed on my machine. PIP leads to this GithHub project as official home page.
However, the releases here (as well as git tags) are only up to version `0.5.4` (as for 13 Dec 2020), pointing this as latest.
I've found commit for `0.6.0-rc1` on `develop`, but the PyPI repository seem to have a regular final `0.6.0` package version available, not just "rc".
Os, where can be sources for the full `0.6.0` version (as released to PyPI) be found?
Thanks,

Obijuan, 2020-12-13:
it is in the master branch
https://github.com/FPGAwars/apio/tree/master

It was releases from master to pypi. But i forgot to tag it on github. I've just created the release on github to avoid confusion
The master branch will be merged into develop and continue de development there

closed 2020-12-13 by Obijuan

achary, 2020-12-15:
Thanks, appreciate that.