FPGAwars/apio #223: Programming TinyFGPA-BX with `apio upload` gives an error.

achary, 2020-12-13:
The `apio upload` does not work out of box for TinyFPGA-BX board giving

```
Error: board TinyFPGA-BX not connected
```

Even when the board is correctly connected, detected and correctly programmable by `tinyprog`.

My environment is:
* Ubuntu 20.04
* Python 3.8
* apgio version 0.6.0
* TinyFGPA-BX board

I've installed `ice40` and `yosys` packages as well as `tinyprog` (separately). I've also executed `apio drivers --serial-enable` without errors.

The board, when in bootloader mode, is mounted as `/dev/ttyACM1` and as a used,  I have the permissions to access it. Also, the `/etc/udev/rules.d/80-fpga-serial.rules` is installed and is picked up by `udev` subsystem correctly.

The workaround to the problem I've found on https://discourse.tinyfpga.com/t/atom-error-board-tinyfpga-bx-not-available/528 posted by `lukevalenty` works, that is, using `tinyprog --pyserial -p hardware.bin` instead of `apio upload` does the job.

I can correctly program the board manually with:

```
tinyprog --pyserial -p hardware.bin
```

as well as simpy:

```
tinyprog -p hardware.bin
```

Would be nice to have `apio upload` work, or have an ability for the user to add `--pyserial` as an extra argument `apio upload` command uses under the hood.

Obijuan, 2020-12-13:
I cannot reproduce your error. apio upload works ok for me with my Tiny:

```
obijuan@Hoth:~/test/TinyFPGA-BX/Blinky$ apio upload
[Sun Dec 13 22:04:49 2020] Processing TinyFPGA-BX
--------------------------------------------------------------------------------
yosys -p "synth_ice40 -json hardware.json" -q Blinky.v
nextpnr-ice40 --lp8k --package cm81 --json hardware.json --asc hardware.asc --pcf TinyFPGA-BX-pins.pcf -q
Warning: unmatched constraint 'PIN_1' (on line 41)
Warning: unmatched constraint 'PIN_2' (on line 42)
Warning: unmatched constraint 'PIN_3' (on line 43)
Warning: unmatched constraint 'PIN_4' (on line 44)
Warning: unmatched constraint 'PIN_5' (on line 45)
Warning: unmatched constraint 'PIN_6' (on line 46)
Warning: unmatched constraint 'PIN_7' (on line 47)
Warning: unmatched constraint 'PIN_8' (on line 48)
Warning: unmatched constraint 'PIN_9' (on line 49)
Warning: unmatched constraint 'PIN_10' (on line 50)
Warning: unmatched constraint 'PIN_11' (on line 51)
Warning: unmatched constraint 'PIN_12' (on line 52)
Warning: unmatched constraint 'PIN_13' (on line 53)
Warning: unmatched constraint 'PIN_14' (on line 56)
Warning: unmatched constraint 'PIN_15' (on line 57)
Warning: unmatched constraint 'PIN_16' (on line 58)
Warning: unmatched constraint 'PIN_17' (on line 59)
Warning: unmatched constraint 'PIN_18' (on line 60)
Warning: unmatched constraint 'PIN_19' (on line 61)
Warning: unmatched constraint 'PIN_20' (on line 62)
Warning: unmatched constraint 'PIN_21' (on line 63)
Warning: unmatched constraint 'PIN_22' (on line 64)
Warning: unmatched constraint 'PIN_23' (on line 65)
Warning: unmatched constraint 'PIN_24' (on line 66)
Warning: unmatched constraint 'SPI_SS' (on line 69)
Warning: unmatched constraint 'SPI_SCK' (on line 70)
Warning: unmatched constraint 'SPI_IO0' (on line 71)
Warning: unmatched constraint 'SPI_IO1' (on line 72)
Warning: unmatched constraint 'SPI_IO2' (on line 73)
Warning: unmatched constraint 'SPI_IO3' (on line 74)
Warning: unmatched constraint 'PIN_25' (on line 77)
Warning: unmatched constraint 'PIN_26' (on line 78)
Warning: unmatched constraint 'PIN_27' (on line 79)
Warning: unmatched constraint 'PIN_28' (on line 80)
Warning: unmatched constraint 'PIN_29' (on line 81)
Warning: unmatched constraint 'PIN_30' (on line 82)
Warning: unmatched constraint 'PIN_31' (on line 83)
Warning: unmatched constraint 'USBP' (on line 89)
Warning: unmatched constraint 'USBN' (on line 90)
39 warnings, 0 errors
icepack hardware.asc hardware.bin
tinyprog --pyserial -c /dev/ttyACM0 --program hardware.bin
Erasing:  97%|█████████▋| 131k/135k [00:00<00:00, 146kB/s]
Erasing: 100%|██████████| 135k/135k [00:00<00:00, 146kB/s]
Writing:  89%|████████▉ | 120k/135k [00:00<00:00, 197kB/s]
Writing: 100%|██████████| 135k/135k [00:00<00:00, 199kB/s]
Reading:  74%|███████▍  | 99.7k/135k [00:00<00:00, 498kB/s]
Reading: 100%|██████████| 135k/135k [00:00<00:00, 496kB/s]

TinyProg CLI
------------
Using device id 1d50:6130
Programming /dev/ttyACM0 with hardware.bin
Programming at addr 028000
Waking up SPI flash
135100 bytes to program
Success!

========================= [SUCCESS] Took 4.11 seconds =========================
```
The tinyprog is executed with the --pyserial option. Check your apio upload output with mine to see if there is something different

Have a look at the [Quick start page](https://github.com/FPGAwars/apio/wiki/Quick-start). Check the apio installation for Ubuntu 20.04 and the  "Testing the TinyFPGA-BX board" sections. It should work out of the box

achary, 2020-12-13:
Interesting. Thanks for posting the logged output.
When I execute `apio upload`, there is no such verbose output, e.g. I do not see the underlying `tinyprog` command that was actually issued. I only get:

```
(.env2) adam@adam-XPS-15-9570:~/workspaces/learning-hdl/tinyfgpa/blink$ apio upload
Activate bootloader by pressing the reset button
Error: board TinyFPGA-BX not connected
```

How could I enforced that verbose output?

achary, 2020-12-13:
BTW: I've just double-checked the recommended "Testing the TinyFPGA-BX board" chapter ending with exactly same error as I've posted.

Obijuan, 2020-12-14:
What I sent to you is the default behaviour. You do not have to activate any verbose output. I've tested that instruction on different computers and they all worked ok. So, the key is finding what is different in your system. Let's start by installing apio from scratch:

1) Remove the .apio folder in your home folder  
2) Install apio on the system (not in a virtualenv by the moment) following the instructions given at: https://github.com/FPGAwars/apio/wiki/Quick-start#apio-installation  
3) Once it is done, check the apio version and the installed apio packages. You should have this output:

```
obijuan@corellia:~$ apio --version
apio, version 0.6.0
obijuan@corellia:~$ apio install -l

Installed packages:

--------------------------------------------------------------------------------
Name        Description                    Version 
--------------------------------------------------------------------------------
dfu         Device Firmware Upgrade Utilities 2020.11.24
ecp5        Lattice ECP5 toolchain         2019.12.11
examples    Verilog examples               0.0.28  
fujprog     Programmer for ULX2/3S boards  2020.10.6
ice40       Lattice ICE40 toolchain        2019.12.11
icesprog    iCESugar board programmer      1.0.0   
iverilog    Icarus Verilog toolchain       1.1.1   
scons       Scons tool                     3.0.1   
system      System tools                   1.1.1   
verilator   Verilator toolchain            1.0.0   
yosys       Yosys toolchain                2019.12.11
obijuan@corellia:~$
```
4) You previously followed the quick start instructions for the TinyFPGA board (so all the configurations should be ok). Just go to the TinyFPGA-BX/Blinky folder. Unplug the TinyFPGA and plug it again (for making sure it is in the boot mode) and execute apio upload

In my case, as the hardware.bin was already generated, apio upload just call tinyprog. This is the output:

```
obijuan@corellia:~/test/TinyFPGA-BX/Blinky$ apio upload
[Mon Dec 14 08:10:34 2020] Processing TinyFPGA-BX
---------------------------------------------------------------------------------
tinyprog --pyserial -c /dev/ttyACM0 --program hardware.bin
Erasing:  97%|█████████▋| 131k/135k [00:00<00:00, 142kB/s]
Erasing: 100%|██████████| 135k/135k [00:00<00:00, 144kB/s]
Writing:  99%|█████████▉| 134k/135k [00:00<00:00, 222kB/s]
Writing: 100%|██████████| 135k/135k [00:00<00:00, 222kB/s]
Reading:  70%|███████   | 95.1k/135k [00:00<00:00, 480kB/s]
Reading: 100%|██████████| 135k/135k [00:00<00:00, 474kB/s]

TinyProg CLI
------------
Using device id 1d50:6130
Programming /dev/ttyACM0 with hardware.bin
Programming at addr 028000
Waking up SPI flash
135100 bytes to program
Success!

/home/obijuan/.apio/packages/tool-scons/script/../engine/SCons/Defaults.py:213: SyntaxWarning: "is not" with a literal. Did you mean "!="?
if len(operation_list) is not 2:
========================== [SUCCESS] Took 2.43 seconds ==========================
```

4) Once the blinky example is uploaded, if you execute apio uload again you will get this message:

```
obijuan@corellia:~/test/TinyFPGA-BX/Blinky$ apio upload
Activate bootloader by pressing the reset button
Error: board TinyFPGA-BX not connected
```
This is because the board is not in boot mode. Sometimes I've found it hard to make the board to enter into the boot mode after uploading the blinky (maybe the example has a bug, I have to check it). Pressing the button sometimes does not work. The safest way to enter into boot mode is to unplug it and plug it again. Just do this way by the moment

Also make sure you do not have any other fpga board already connected (like the fomu)

Obijuan, 2020-12-14:
ups... i've close it by mistake. It is reopen again

achary, 2020-12-14:
This is very odd, my default behaviour is very different than yours. I do not see the actual commands `apio upload` (v0.6.0) uses printed to stdout. This drives me crazy at this point.
I definitely enter boot-loader mode before any bitstram upload attempt. At each time - getting the error. And at each point, I can successfully work around and program with direct execution of `tinyprog` command.

I've followed your advice and I did installed `apio` and `tinyprog` in the system, per the instructions from you link.
That made no difference whatsoever.

Here is my output:

```
adam@adam-XPS-15-9570:~/workspaces/learning-hdl/tinyfgpa/blink$ apio --version
apio, version 0.6.0
adam@adam-XPS-15-9570:~/workspaces/learning-hdl/tinyfgpa/blink$ apio install -l

Installed packages:

---------------------------------------------------------------------------------------------------------------------------------------
Name        Description                    Version 
---------------------------------------------------------------------------------------------------------------------------------------
dfu         Device Firmware Upgrade Utilities 2020.11.24
ecp5        Lattice ECP5 toolchain         2019.12.11
examples    Verilog examples               0.0.28  
fujprog     Programmer for ULX2/3S boards  2020.10.6
ice40       Lattice ICE40 toolchain        2019.12.11
icesprog    iCESugar board programmer      1.0.0   
iverilog    Icarus Verilog toolchain       1.1.1   
scons       Scons tool                     3.0.1   
system      System tools                   1.1.1   
verilator   Verilator toolchain            1.0.0   
yosys       Yosys toolchain                2019.12.11

adam@adam-XPS-15-9570:~/workspaces/learning-hdl/tinyfgpa/blink$ apio upload
Activate bootloader by pressing the reset button
Error: board TinyFPGA-BX not connected
```

achary, 2020-12-14:
I've also noticed that I have two serial devices that apio sees:

```
adam@adam-XPS-15-9570:~/workspaces/TinyFPGA-BX/Blinky$ apio system --lsserial
Number of Serial devices found: 2

/dev/ttyACM1
Description: ttyACM1
Hardware info: USB VID:PID=1209:2100 LOCATION=1-5.4:1.0

/dev/ttyACM0
Description: Goodix Fingerprint Device
Hardware info: USB VID:PID=27C6:5395 SER=HTK32 LOCATION=1-7:1.0
```

The right one is `/dev/ttyACM1` in my case. Could this be related to how apio functions, like, maybe there is something that apio is confused when two devices are present at a time?

I've tried to run `apio upload --serial-port /dev/ttyACM1` with no effect, though.

Obijuan, 2020-12-14:
Yes! I think that could be the problem. In my case I only have one serial port:

```
obijuan@corellia:~/test/TinyFPGA-BX/Blinky$ apio system --lsserial
Number of Serial devices found: 1

/dev/ttyACM0
Description: ttyACM0
Hardware info: USB VID:PID=1D50:6130 LOCATION=2-7.1.3:1.0
```
I guest that as you said, apio get confused. If I can reproduce the bug with my board, i would be able to fix it

In the meanwhile you can try the apio raw command (it just executes directly the binaries already installed)

```
obijuan@corellia:~/test/TinyFPGA-BX/Blinky$ apio raw "tinyprog -l"

    TinyProg CLI
    ------------
    Using device id 1d50:6130
    Only one board with active bootloader, using it.
    Boards with active bootloaders:

        /dev/ttyACM0: TinyFPGA BX 1.0.0
            UUID: 7d41d659-876b-454a-9a91-51e5f157e80c
            FPGA: ice40lp8k-cm81

obijuan@corellia:~/test/TinyFPGA-BX/Blinky$
```

achary, 2020-12-14:
Thanks.
BTW: I've noticed that connecting Arduino yields another `/dev/ttyACMx` port that `apio` also lists with `apt system --lsserial` command, so if you happen to have one, that might a way to potentially reproduce the issue.

achary, 2020-12-15:
I was able to find where the issue was, and the good news is `apio` works fine,
and it was my board to put the blame on.

Here is what I was able to find, for a reference.

When I plugged my TinyFPGA-BX device to the USB socket, the `dmesg` reported it as:

```
...
[596070.364050] usb 1-5.4: New USB device found, idVendor=1209, idProduct=2100, bcdDevice= 0.00
...
```

I was able to find that when `apio` tried to identify which programmer to use
(in `scons.py` file, `check_usb` method), it correctly had this device in
available USB devices list: (as of `System().get_usb_devices()`):

```
[{'hwid': '0cf3:e300'}, {'hwid': '413c:2110'}, {'hwid': '03f0:0941'}, {'hwid': '1d6b:0003'}, {'hwid': '05e3:0749'}, {'hwid': '1209:2100'}, {'hwid': '0c45:671d'}, {'hwid': '27c6:5395'}, {'hwid': '413c:1010'}, {'hwid': '05e3:0610'}, {'hwid': '05e3:0610'}, {'hwid': '1d6b:0002'}, {'hwid': '1d6b:0002'}, {'hwid': '0bda:8153'}, {'hwid': '05e3:0626'}, {'hwid': '05e3:0626'}, {'hwid': '1d6b:0003'}]
```

However, this particular line in `check_usb` method (`scons.py:233`):

```
usb_data = board_data.get('usb')
```

returned the following dict:

```
{'name': 'TinyFPGA BX', 'fpga': 'iCE40-LP8K-CM81', 'programmer': {'type': 'tinyprog'}, 'usb': {'vid': '1d50', 'pid': '6130'}, 'tinyprog': {'desc': 'TinyFPGA BX'}}
```

and the VID:PID pair was different that the one `dmesg`
reported, but also invalid in general, as I had no such device in the
system.

It looks like those VID:PID for the programmer were defined in `boards.json`:

```
...
  "TinyFPGA-BX": {
    "name": "TinyFPGA BX",
    "fpga": "iCE40-LP8K-CM81",
    "programmer": {
      "type": "tinyprog"
    },
    "usb": {
      "vid": "1d50",
      "pid": "6130"
    },
    "tinyprog": {
      "desc": "TinyFPGA BX"
    }
  },
...
```

The board I've acquired recently had a different VID:PID (as mentioned,
they were `1209:2100`). And this is what made our environments different, and
why you could not reproduce my issue :).

Perhaps if I did start from looking into what `lsub` command reported, I would have been able
to spot that it was:

```
...
Bus 001 Device 074: ID 1209:2100 Generic TinyFPGA B1 and B2 Boards
...
```
[... 18 lines cut ...]
        New Version: 1.0.1                                                                                                               
        Notes: Updates USB VID:PID to fix issues with APIO not recognizing the board.                                                                                                                                                                                             
                                                                                                                                                                                                                                                                                  
    is available for this board:                                                                                                                                                                                                                                                  
                                                                                                                                                                                                                                                                                  
        /dev/ttyACM1: TinyFPGA BX 1.0.0                                                                                                                                                                                                                                           
            UUID: blahblahblah-blah...                                                                                                                                                                                                                           
            FPGA: ice40lp8k-cm81                                                                                                                                                                                                                                                  
                                                                                                                                                                                                                                                                                  
    Would you like to perform the update? [y/N] y   
```

After that `apio upload` worked **exactly as expected**.

I think the lesson from here is that the error message reported did not
immediately lead to possible cause of a problem (even `apio` potentially had
the information to link the two facts).

I would recommend to have a clear warning in the Wiki pages for TinyFGPA-BX
board asking for explicitly double-checking the VID:PID the possessed board
reports, and updating the bootloader if a mismatch is there. I'm sure there are still
TinyFGPA-BXs with inappropriate bootloader variant out there in the field,
potentially impacted by this issue.

Anyway, thanks for helping in figuring out this case.

achary, 2020-12-15:
I've added an additional check point on Wiki page for it, so I think no further actions are required.

closed 2020-12-15 by achary