FPGAwars/apio #224: apio sim and apio verify not working if the home path contains spaces

Obijuan, 2020-12-15:
Bug reported by Jorge Garcia Mateos on the FPGAwar group: [Preguntas de novato](https://groups.google.com/g/fpga-wars-explorando-el-lado-libre/c/AeDyzrN8VW8) (in Spanish)

```
apio sim
iverilog -o leds_tb.out -D VCD_OUTPUT=leds_tb C:/Users/Jorge G/.apio\packages\toolchain-yosys\share\yosys/ice40/cells_sim.v leds.v leds_tb.v
C:/Users/Jorge: No such file or directory
No top level modules, and no -s option.
scons: *** [leds_tb.out] Error 1
```

alexachso, 2021-11-25:
bump

Still the case. Can't we specify the path where the install command downloads the packages?

Obijuan, 2022-05-01:
This bug should have been fixed in the latest apio 0.8.1. Please try it and feel free to reopen it if the problem remain. Or create a new issue if new bugs are found
Thanks for the feedback

closed 2022-05-01 by Obijuan