FPGAwars/apio #227: Manage code with subdirectories

nihirash, 2021-03-27:
Hello!

Is it possible use APIO with project that didn't organized as a single directory?
I'd like have directory list, where my verilog files placed and run apio build for it. 

BTW, thank you for your great tool!

nihirash, 2021-04-10:
Currently did with "raw" call of utils.

pm100, 2024-04-05:
I think my ask is similar. I would like to have a library of modules that I can include in many projects. Icecube2 allows this. This is probably a request to yosys, but I am very new to these tools.

I dont want to use `include

zapta, 2025-03-22:
This PR adds support for subdirectores. https://github.com/FPGAwars/apio/pulls

Apio is agnostic to the subdirectories and user can organize the source files any way they want. 

This example demonstrates a project with sub directores.  https://github.com/zapta/apio-examples/tree/master/examples/alhambra-ii/bcd-counter

Closing.

closed 2025-03-22 by zapta