FPGAwars/apio #236: Tests fail

Luflosi, 2021-07-20:
Trying to execute the test suite on the latest version 0.7.4 leads to a failure:
```
============================= test session starts ==============================
platform darwin -- Python 3.8.9, pytest-6.2.4, py-1.10.0, pluggy-0.13.1
rootdir: /private/tmp/nix-build-apio-0.7.4.drv-0/source
collected 50 items                                                             

test/test_apio.py ..                                                     [  4%]
test/code_commands/test_build.py ....                                    [ 12%]
test/code_commands/test_clean.py .                                       [ 14%]
test/code_commands/test_lint.py .                                        [ 16%]
test/code_commands/test_sim.py .                                         [ 18%]
test/code_commands/test_time.py ..                                       [ 22%]
test/code_commands/test_upload.py ........                               [ 38%]
test/code_commands/test_verify.py .                                      [ 40%]
test/env_commands/test_boards.py ...                                     [ 46%]
test/env_commands/test_config.py ...                                     [ 52%]
test/env_commands/test_drivers.py .                                      [ 54%]
test/env_commands/test_examples.py ....                                  [ 62%]
test/env_commands/test_init.py .....                                     [ 72%]
test/env_commands/test_install.py ...                                    [ 78%]
test/env_commands/test_system.py .FF..                                   [ 88%]
test/env_commands/test_uninstall.py ...                                  [ 94%]
test/env_commands/test_upgrade.py s                                      [ 96%]
test/packages/test_api.py s                                              [ 98%]
test/packages/test_complete.py s                                         [100%]

=================================== FAILURES ===================================
______________________________ test_system_lsftdi ______________________________

clirunner = <click.testing.CliRunner object at 0x10c66b2e0>
configenv = <function configenv.<locals>.decorator at 0x10c63cdc0>

    def test_system_lsftdi(clirunner, configenv):
        with clirunner.isolated_filesystem():
            configenv()
            result = clirunner.invoke(cmd_system, ['--lsftdi'])
            assert result.exit_code == 1
>           assert 'apio install system' in result.output
E           assert 'apio install system' in "DEBUG: APIO_HOME_DIR: /private/tmp/nix-build-apio-0.7.4.drv-0/tmpuzchl07z/ ñ\nProfile path: /private/tmp/nix-build-ap...bin/lsftdi\nERROR!!!!!!\nError: package 'oss-cad-suite' is not installed\nPlease run:\n   apio install oss-cad-suite\n"
E            +  where "DEBUG: APIO_HOME_DIR: /private/tmp/nix-build-apio-0.7.4.drv-0/tmpuzchl07z/ ñ\nProfile path: /private/tmp/nix-build-ap...bin/lsftdi\nERROR!!!!!!\nError: package 'oss-cad-suite' is not installed\nPlease run:\n   apio install oss-cad-suite\n" = <Result SystemExit(1)>.output

test/env_commands/test_system.py:14: AssertionError
______________________________ test_system_lsusb _______________________________

clirunner = <click.testing.CliRunner object at 0x10c66b2e0>
configenv = <function configenv.<locals>.decorator at 0x10c63cdc0>

    def test_system_lsusb(clirunner, configenv):
        with clirunner.isolated_filesystem():
            configenv()
            result = clirunner.invoke(cmd_system, ['--lsusb'])
            assert result.exit_code == 1
>           assert 'apio install system' in result.output
E           assert 'apio install system' in "DEBUG: APIO_HOME_DIR: /private/tmp/nix-build-apio-0.7.4.drv-0/tmpogwgpv9d/ ñ\nProfile path: /private/tmp/nix-build-ap... bin/lsusb\nERROR!!!!!!\nError: package 'oss-cad-suite' is not installed\nPlease run:\n   apio install oss-cad-suite\n"
E            +  where "DEBUG: APIO_HOME_DIR: /private/tmp/nix-build-apio-0.7.4.drv-0/tmpogwgpv9d/ ñ\nProfile path: /private/tmp/nix-build-ap... bin/lsusb\nERROR!!!!!!\nError: package 'oss-cad-suite' is not installed\nPlease run:\n   apio install oss-cad-suite\n" = <Result SystemExit(1)>.output

test/env_commands/test_system.py:22: AssertionError
=========================== short test summary info ============================
FAILED test/env_commands/test_system.py::test_system_lsftdi - assert 'apio in...
FAILED test/env_commands/test_system.py::test_system_lsusb - assert 'apio ins...
=================== 2 failed, 45 passed, 3 skipped in 1.78s ====================
```
Version 0.6.7 worked fine.
I'm not 100% sure if this is a problem with apio or with my unusual build environment. Let me know if you can reproduce the test failure.
I'm happy to provide any additional information that may be useful.

Obijuan, 2021-07-20:
It was my fault. I've changed some commands but not updated the tests  
It should be fixed in the develop branch. Could you please try it agan?

Obijuan, 2021-07-20:
This is my current output when running the tests:
```
(venv) obijuan@corellia:~/Develop/FPGAwars/apio$ make tests
pytest apio test
=============================================== test session starts ===============================================
platform linux -- Python 3.8.10, pytest-6.2.4, py-1.10.0, pluggy-0.13.1
rootdir: /home/obijuan/Develop/FPGAwars/apio
collected 50 items                                                                                                

test/test_apio.py ..                                                                                        [  4%]
test/code_commands/test_build.py ....                                                                       [ 12%]
test/code_commands/test_clean.py .                                                                          [ 14%]
test/code_commands/test_lint.py .                                                                           [ 16%]
test/code_commands/test_sim.py .                                                                            [ 18%]
test/code_commands/test_time.py ..                                                                          [ 22%]
test/code_commands/test_upload.py ........                                                                  [ 38%]
test/code_commands/test_verify.py .                                                                         [ 40%]
test/env_commands/test_boards.py ...                                                                        [ 46%]
test/env_commands/test_config.py ...                                                                        [ 52%]
test/env_commands/test_drivers.py .                                                                         [ 54%]
test/env_commands/test_examples.py ....                                                                     [ 62%]
test/env_commands/test_init.py .....                                                                        [ 72%]
test/env_commands/test_install.py ...                                                                       [ 78%]
test/env_commands/test_system.py .....                                                                      [ 88%]
test/env_commands/test_uninstall.py ...                                                                     [ 94%]
test/env_commands/test_upgrade.py .                                                                         [ 96%]
test/packages/test_api.py .                                                                                 [ 98%]
test/packages/test_complete.py .                                                                            [100%]

=============================================== 50 passed in 2.92s ================================================
(venv) obijuan@corellia:~/Develop/FPGAwars/apio$
```

Luflosi, 2021-07-20:
Yep, all good now. Thank you.

closed 2021-07-20 by Luflosi