FPGAwars/apio #238: Unable to build for TinyFPGA-BX

matt-hu, 2021-08-02:
I am unable to build for the TinyFPGA-BX. Running `apio build --board TinyFPGA-BX` gives this error:
```
/home/matt/.apio/packages/tools-oss-cad-suite/libexec/nextpnr-ice40: /home/matt/.apio/packages/tools-oss-cad-suite/lib/libc.so.6: version `GLIBC_2.32' not found (required by /lib/libpython3.8.so.1.0)
/home/matt/.apio/packages/tools-oss-cad-suite/libexec/nextpnr-ice40: /home/matt/.apio/packages/tools-oss-cad-suite/lib/libc.so.6: version `GLIBC_2.33' not found (required by /lib/libpython3.8.so.1.0)
```
I am on Arch Linux, using APIO installed from pip. I have glibc and lib32-glibc installed, and I am using Python 3.8 (the system default was 3.9, but this was giving me errors about being unable to find something to do with 3.8, so I installed it then reinstalled apio under 3.8)
I have also tried building nextpnr from source, this works, as does using ~/.apio/packages/tools-oss-cad-suite/libexec/nextpnr-ice40, so I am unsure what is causing the problem. I am assuming it is something to do with apio?

AlienTux, 2021-08-14:
I have the same issue in Solus Gnome.
I also have the libs installed, but version 2.3.3. Should I install 2.3.2 alongside 2.3.3?

Obijuan, 2021-08-14:
Yes, we are currently upgrading the toolchains to their latest versions. During the next couple of weeks there could be broken dependencies. Sorry for the inconvenience

AlienTux, 2021-08-14:
Understood! Thank you for all your hard work!

Obijuan, 2022-05-01:
I thik this bug has been fixed. Please try the latest release (apio 0.8.1). Feel free to reopen this issue if the problem remain. Or create a new issue if new bugs are detected

closed 2022-05-01 by Obijuan