FPGAwars/apio #241: SConstruct file does not find package binaries correctly on Windows

kanerogers, 2021-08-08:
Hey there!

Firstly, thanks so much for making such an amazing library. Really looking forward to mucking about with it.

I found the installation process really easy, following along with the great [getting started guide](https://apiodoc.readthedocs.io/en/latest/source/quick_start.html), but ran into troubles when running `apio upload`:

```
PS C:\Users\kanem\Development\learn-fgpa\leds> apio upload
(DEBUG) Profile path: C:\Users\kanem\.apio\profile.json
(DEBUG) Home_dir: C:\Users\kanem\.apio
(DEBUG) Profile path: C:\Users\kanem\.apio\profile.json
(DEBUG) Home_dir: C:\Users\kanem\.apio
(DEBUG) Run Command: lsusb
(DEBUG) System_base_dir: C:\Users\kanem\.apio\packages\tools-oss-cad-suite
(DEBUG) System bin dir: C:\Users\kanem\.apio\packages\tools-oss-cad-suite\bin
(DEBUG) Executable file: C:\Users\kanem\.apio\packages\tools-oss-cad-suite\bin\lsusb.exe
(DEBUG) Profile path: C:\Users\kanem\.apio\profile.json
(DEBUG) Home_dir: C:\Users\kanem\.apio
(DEBUG) Run Command: lsftdi
(DEBUG) System_base_dir: C:\Users\kanem\.apio\packages\tools-oss-cad-suite
(DEBUG) System bin dir: C:\Users\kanem\.apio\packages\tools-oss-cad-suite\bin
(DEBUG) Executable file: C:\Users\kanem\.apio\packages\tools-oss-cad-suite\bin\lsftdi.exe
[Sun Aug  8 19:25:20 2021] Processing icestick
-------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------
Executing: c:\python39\python.exe C:\Users\kanem\.apio\packages\tool-scons\script\scons -Q upload fpga_arch=ice40 fpga_size=1k fpga_type=hx fpga_pack=tq144 prog=iceprog -d i:0x0403:0x6010:0 -f c:\python39\lib\site-packages\apio\resources\ice40\SConstruct
C:\Users\kanem\.apio\packages\toolchain-yosys\share\yosys
yosys -p "synth_ice40 -json hardware.json" leds.v
scons: *** [hardware.json] Error 3221225781
=============================================================================== [ ERROR ] Took 0.32 seconds ===============================================================================
PS C:\Users\kanem\Development\learn-fgpa\leds>
```

After poking around a bit, it seems like the problem is on this line in `SConstruct`:

```
synth = Builder(
    action='yosys -p \"synth_ice40 -json $TARGET\" {} $SOURCES'.format(
                                                                    '' if VERBOSE_ALL or VERBOSE_YOSYS else '-q'
                                                                    ),
    suffix='.json',
    src_suffix='.v',
    source_scanner=list_scanner
```

I've worked around this by just doing:

```
# -- HACK
YOSYS_BIN = "C:\\Users\\kanem\\.apio\\packages\\toolchain-yosys\\bin\\yosys.exe"
synth = Builder(
    action='{} -p \"synth_ice40 -json $TARGET\" {} $SOURCES'.format(YOSYS_BIN,
        '' if VERBOSE_ALL or VERBOSE_YOSYS else '-q'
    ),
```

And so on for the rest of the tools.

Now, I saw that `util.py` has a function `get_bin_dir_table` that could be use to portably get the correct package binary. I'd be happy to implement a fix that passes the path of each of these binaries into `SConstruct` as a variable, if you're interested! Otherwise, if I'm just doing something stupid or there's a better fix available, I'll just work around this for now. 😄 

Thanks again!

kanerogers, 2021-08-31:
@Obijuan any thoughts on this one? I saw the toolchain is being updated soon, is that right? 😄

Obijuan, 2021-08-31:
Yes, the toolchain is being updated, so currently there are many broken stuff. After the updating we have to find and fix the bugs. Please, keep this issue open until it can be tested with the updated toolchain

kanerogers, 2021-08-31:
Not a problem, thanks so much!! Happy to be a guinea pig for Windows. ❤️

quizque, 2021-12-13:
Just want to add to this since it's a very similar issue. The command `apio verify` doesn't work when you have a space in your username on windows. 
```
PS C:\dev\fpga\UPduino-v3.0\RTL\blink_led> apio verify
(DEBUG) Profile path: C:\Users\Nick [removed]\.apio\profile.json
(DEBUG) Home_dir: C:\Users\Nick [removed]\.apio
iverilog -o hardware.out -D VCD_OUTPUT= C:\Users\Nick [removed]\.apio\packages\toolchain-yosys\share\yosys/ice40/cells_sim.v rgb_blink.v
C:\Users\Nick: No such file or directory
No top level modules, and no -s option.
scons: *** [hardware.out] Error 1
======================================= [ ERROR ] Took 0.41 seconds =======================================
```

nmaas87, 2022-02-02:
Some problem as @quizque 
````
iverilog -o hardware.out -D VCD_OUTPUT= -D NO_ICE40_DEFAULT_ASSIGNMENTS C:\Users\Nico Maas\.apio\packages\tools-oss-cad-suite\share\yosys/ice40/cells_sim.v leds.v
C:\Users\Nico: No such file or directory
No top level modules, and no -s option.
scons: *** [hardware.out] Error 1
============================================= [ ERROR ] Took 0.87 seconds =============================================
PS C:\Users\Nico Maas\Documents\apio\icestick\leds> apio verify
````

Under Windows 10, username with blank does not allow for apio verify. Any idea on how to install apio e.g. into a non user directory, like C:\apio ?

quizque, 2022-02-02:
@nmaas87 The only work around I have found was by following the fix in this issue https://github.com/FPGAwars/apio/issues/254#issuecomment-973411855 then fully removing anything APIO related (I don't remeber the exact folders, but I found them using Everything and search "APIO"). After that, add an environment variable `APIO_HOME_DIR ` pointing to `C:/apio` and then install apio again. It take a while but I hope this helps.

nmaas87, 2022-02-02:
Thank you @quizque - that worked like a charm! I just uninstalled with ````apio unistall --all````, create the variable, re-opened my Admin Powershell and run ````apio install --all````. It directly installed to C:\apio and I had no issues anymore 👍🏻 
Maybe documenting such a powerful function in the Wiki or Quick Install?
Thank you!

Obijuan, 2022-05-01:
A new apio 0.8.1 has been release. The migration of toolchains has been completed. This error should be fixed. Please, try it and feel free to reopen this issue in case they remain. Or create a new issue if new bugs are found
Thanks a lot for your feedback

closed 2022-05-01 by Obijuan