FPGAwars/apio #242: Error at apio build on Windows10

hpsteidle, 2021-08-17:
I installed apio 0.7.6 and tested the toolchain with the example "leds" on Windows10. "apio verify" and "apio sim" is working. When i try to use "apio build", there comes a System-Error Message from yosys.exe: "..libreadline8.dll not found..". On the Console, there is another messsage "scons: *** [hardware.json] Error 3221225781".

I did the same testprocedure with the Toolchains 0.6.4 an 0.6.7. There worked  everything fine.

Obijuan, 2021-08-22:
Currently the toolchains are being updated, so during the next weeks there will be many errors. I apologize for the inconvenience

ShawnHymel, 2021-10-06:
@Obijuan any chance you could provide an estimate on when those updates might be done? I'm currently working on an "Intro to FPGA" series for Digi-Key's YouTube channel, and I think apio would be the perfect tool for beginners.

Obijuan, 2022-05-01:
The toolchains have already been updated. It should work ok with apio 0.8.1. Please try it and feel free to reopen this issue in case it remains (or open a new issue if new errors are detected)
Thanks a lot for the feedback

closed 2022-05-01 by Obijuan