FPGAwars/apio #245: Error code 404 when trying to install ice40 package

jevonlongdell, 2021-09-21:
When I run `apio install ice40` I get the following. Error message below.

It seems is confused about what tar.gz to download and tries to get one that doesn't exist.

I'd really appreciate any help available with this issue. I was all set up to use these icestudio in a course I'm teaching and this problem has me in a bit of a bind.

-----------------------------------------

(DEBUG) Profile path: C:\Users\Phsi245\.apio\profile.json
(DEBUG) Home_dir: C:\Users\Phsi245\.apio
Installing ice40 package:
Warning: full platform does not match: windows_amd64
         Trying OS name: windows
Error: Got an unrecognized status code '404' when downloaded https://github.com/FPGAwars/toolchain-ice40/releases/download/v2019.12.11/toolchain-ice40-windows-2019.12.11.tar.gz

Obijuan, 2021-09-21:
Please, make sure you are using apio 0.76 (it is the current stable version)  
You can check it with the command: `apio --version`

With version 0.76 it works ok on the windows platforms I've tested  
If it still is not working for you we should find what is different in your system

Anyway, Icestudio install its own apio in a python virtualenv (independly of the apio that you have in your system), so there is no need to use the external apio (the installation and building of circuits is transparent for the user).  Is it working ok icestudio with the hello-world circuit?

jevonlongdell, 2021-09-24:
Thanks for yor reply. I tried it again and it worked. Not sure what the issue was. 

BTW I installed apio outside icestudio because icestudio was complaining about ice40 not being installed, eventhough it didnt flag any errors when installing the toolchain. The same issue was evidently effecting both python environments, who knows what it was.

jevonlongdell, 2021-09-24:
the version of apio was whatever pip gave me.

jevonlongdell, 2021-09-24:
Im not sure if I can close this myself or not, but I think this issue should be closed.

Thanks again.

closed 2022-04-24 by Obijuan