FPGAwars/apio #248: TinyFPGA A-Series support

aelray, 2021-09-28:
Hello,

ist it possibile setup APIO to work with the A-Series TinyFPGAs? 

They use a MachXO2-256  or a MachXO2-1200.

Any help would be appreciated.

Regards,
Adel

zapta, 2024-12-23:
Looking at ``~/.apio/packages/tools-oss-cad-suite/bin``, apio already install this binary ``nextpnr-machxo2``.  If this architecture we want to support need to add a new SConstruct file.

zapta, 2024-12-23:
@Obijuan, this is classified as a 'board request'. You can create a new category for 'architecture request', which is obviously more involved.

cavearr, 2026-09-16:
@zapta @Obijuan 

Following the rule you set in January (close if still unavailable by June 2026): the Crowd Supply product page for the TinyFPGA A-Series now returns 404, and the boards have not been purchasable for years. 

Proposing to close. If they ever come back, the MachXO2 side is cheap: the device data already rides in trellis's `devices.json`, and the parts-index generator (#1002) will emit a `MACHXO2-PARTS-INDEX.json` for free, so the catalog would be ready before the boards are.

Obijuan, 2026-09-16:
I agree. My vote if for closing it

zapta, 2026-09-16:
+1 for closing.

I don't see much activity with MachXO2.

closed 2026-10-18 by zapta