FPGAwars/apio #249: Error when apio build on macOS

xralphack, 2021-09-28:
dyld: Library not loaded: @executable_path/../lib/libncurses.6.dylib
Referenced from: /Users/xxx/.apio/packages/tools-oss-cad-suite/lib/libreadline.8.1.dylib
Reason: image not found

jfoucher, 2021-10-15:
Same here, did you find a solution ?

jfoucher, 2021-10-15:
Just leaving this here if anyone needs it, but I fixed it with this : 
```
cp /usr/local/Cellar/ncurses/6.2/lib/libncurses.6.dylib /Users/XXXX/.apio/packages/tools-oss-cad-suite/lib/
```

xralphack, 2021-10-16:
I am currently giving up apio.

kattaliraees, 2021-11-29:
Replacing the entire `tools-oss-cad-suite` folder with the one downloaded from https://github.com/YosysHQ/oss-cad-suite-build/releases worked for me on an M1 Mac with macOS Monterey. `apio sim` didn't work though

Obijuan, 2022-05-01:
This bug has been fixed in the latest apio 0.8.1.  Please try it and feel free to reopen this issue in case the error remains (or create a new issue if new bugs are detected)
Thanks for your feedback

closed 2022-05-01 by Obijuan