FPGAwars/apio #285: iCE40-LP1K-CB121 support

angelsix, 2022-02-06:
I have a board using the LP1K-CB121 and would like to use apio to compile it. It is a custom board for a specific project, but I believe it only matters about the chip itself in order to add support?

angelsix, 2022-02-06:
Forgot to mention I program it using the Lattice HW-USBN-2B directly. This programmer shows in the `apio system -lsftdi` so presume thats also supported

zapta, 2024-11-12:
That FPGA seems to be supported. 

https://github.com/FPGAwars/apio/blob/303dbddce93daf9eb0c7794a4a9391d6bae7e7fc/apio/resources/fpgas.json#L110

The next release of apio will include support for custom boards defined in the project itself and I believe that it will address this request.  This feature is already implemented in the dev version and I am using it myself.

zapta, 2024-11-12:
Closing. Please reopen if you have further questions or comments.

cavearr, 2025-02-04:
@zapta i'm helping an icestudio user that are integrating his own fpga board based on this fpga (iCE40-LP1K-CB121)

we are trying to use the custom board file in the project folder if i use boards.jsonc and call lint:

```
apio lint

Loading custom 'boards.jsonc'.

Error: construct_scons_params(): unknown fpga iCE40-LP1K-CB121

```

The content of boards.jsonc are:

```
// Apio boards definitions.
//
{
  "iCE40-VGA-IPS": {
    "name": "iCE40-VGA-IPS IPS Screen",
    "fpga": "iCE40-LP1K-CB121",
    "programmer": {
      "type": "iceprog"
    },
    "usb": {
      "vid": "0403",
      "pid": "6010"
    },
    "ftdi": {
      "desc": "(?:Dual RS2232H)|(?:Lattice HW-USBN-2B)"
    }
  }
}
```

Where is the error? thanks a lot @zapta !

cavearr, 2025-02-04:
thanks a lot @zapta ! i'll documented the experience to have all very clear, thanks for the support!

zapta, 2025-02-04:
I will add more information in the error message.

zapta, 2025-05-07:
It seems that the issue is resolved. Closing.

closed 2025-05-07 by zapta